microsoft/AirSim #1211: Does time of day affect shadows?

NextSim, 2018-06-28:
How does the time of day setting work?

In the blocks environment, it seems that if TimeOfDay is enabled, regardless of the time used, it looks like it is during sunrise.

In the prebuilt AirSimNH environment I don't see any affect for this setting.

Does it also affect the shadows of objects? Or only the look of the sky?

NextSim, 2018-08-03:
Does anyone know?  

We're trying to train a car to drive and the shadows are causing issues.

bmalhigh, 2018-11-27:
It is a bug. Same as issue https://github.com/Microsoft/AirSim/issues/956
Working on a fix.

bmalhigh, 2018-11-28:
This should be fixed now with the pull request 1581.

TimeOfDay currently impacts the position of the sun (which should impact the shadows indirectly)

closed 2018-11-28 by bmalhigh