microsoft/AirSim #1215: Compile Error - latest Master

CaptainJL, 2018-06-30:
After compilation of Airsim and going through UE4Editor and running the compatibility rebuild (as stated in the docs), i get this error:

```
AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/api/VehicleApiBase.hpp:87:29: error: 
      default initialization of an object of const type 'const
      msr::airlib::RCData' without a user-provided default constructor
        static const RCData invalid_rc_data;
                            ^
                                           {}
```
This is happening using the Linux build on the updated Ubuntu 16.04.
I followed all the steps as stated, and I didnt have this issue a couple weeks ago (I wanted to update Airsim, so i redownloaded it).

As such, the compatibility rebuild is not working.

Any solutions?

mitairlines, 2018-06-30:
I have the same problem.

mitairlines, 2018-07-02:
Thank you for your response,
I tried what you suggested above and now I am getting the following:
The project could not be compiled. Would you like to open it in Null Source Code Access?
`ERROR: UBT ERROR: Failed to produce item: /home/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so
Total build time: 6341.95 seconds (Local executor: 0.00 seconds)`

mitairlines, 2018-07-04:
Thanks so much for your help! It looks like it worked, and blocks compiled properly.  However, now, when I try to run hello_drone.py it says 
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "hello_drone.py", line 12, in <module>
    client.enableApiControl(True)
  File "/home/tanmatthew/Aurora/AirSim/PythonClient/airsim/client.py", line 37, in enableApiControl
    return self.client.call('enableApiControl', is_enabled, vehicle_name)
  File "/home/tanmatthew/.local/lib/python3.5/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/tanmatthew/.local/lib/python3.5/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for '' is not available. This could either because this is simulation-only API or this vehicle does not exist.

stanleycelestin1, 2018-07-05:
It seems to be that you have to package it before you can use the API's on it. You should be able to package in your unreal editor: **File > Package Project > Linux **> choose directory ... open.

stanleycelestin1, 2018-07-05:
Also are you using python2 or  python3?

mitairlines, 2018-07-05:
Thank you for your reply,
I am using python 3.5.
I performed the operation you detailed, but the program still threw the same error:

UATHelper: Packaging (Linux):   error: exception handling was disabled in PCH file but is currently enabled
UATHelper: Packaging (Linux):   error: definition of macro 'PLATFORM_EXCEPTIONS_DISABLED' differs between the precompiled header ('1') and the command line ('0')

UATHelper: Packaging (Linux):   ERROR: UBT ERROR: Failed to produce item: /home/AirSim/Unreal/Environments/Blocks/Binaries/Linux/Blocks
PackagingResults: Error: UBT ERROR: Failed to produce item: /home/AirSim/Unreal/Environments/Blocks/Binaries/Linux/Blocks

ssokhey, 2018-09-03:
@stanleycelestin1Your suggestion works for me. Thanks.

madratman, 2019-01-22:
@mitairlines this is probably too late, but you can get around this by setting `bUsePchFiles=false` in your `PROJECT/Source/Project.target.cs` file. 
Also, the blocks linux binary is now available : https://github.com/Microsoft/AirSim/releases/tag/v1.2.0Linux

closed 2019-01-22 by madratman