microsoft/AirSim #1217: No rule to make target 'cxx'

yonhdee, 2018-06-30:
I'm using Ubuntu16.04 and having these troubles, my ubuntu is newly installed and I haven't modified anything, so grateful if anyone can help

```bash
~/airsim/AirSim  master ✔                                                                                                                                                    2d
▶ ./setup.sh
+ set -e
+++ dirname ./setup.sh
++ cd .
++ pwd
+ SCRIPT_DIR=/home/yonhdee/airsim/AirSim
+ pushd /home/yonhdee/airsim/AirSim
+ downloadHighPolySuv=true
+ [[ '' == \-\-\n\o\-\f\u\l\l\-\p\o\l\y\-\c\a\r ]]
++ uname
+ '[' Linux == Darwin ']'
+ [[ ! -z '' ]]
+ sudo apt-get install -y build-essential
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ wget -O - http://apt.llvm.org/llvm-snapshot.gpg.key
+ sudo apt-key add -
--2018-06-30 21:14:02--  http://apt.llvm.org/llvm-snapshot.gpg.key
Resolving apt.llvm.org (apt.llvm.org)... 151.101.26.49
Connecting to apt.llvm.org (apt.llvm.org)|151.101.26.49|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3145 (3.1K) [application/octet-stream]
Saving to: ‘STDOUT’

-                                                  100%[================================================================================================================>]   3.07K  --.-KB/s    in 0s

2018-06-30 21:14:02 (228 MB/s) - written to stdout [3145/3145]

OK
+ sudo apt-get update
Hit:1 http://mirrors.tuna.tsinghua.edu.cn/ubuntu xenial InRelease
Hit:2 http://mirrors.tuna.tsinghua.edu.cn/ubuntu xenial-updates InRelease
Hit:3 http://mirrors.tuna.tsinghua.edu.cn/ubuntu xenial-backports InRelease
Hit:4 http://mirrors.tuna.tsinghua.edu.cn/ubuntu xenial-security InRelease
Hit:5 https://mirrors.tuna.tsinghua.edu.cn/docker-ce/linux/ubuntu xenial InRelease
Hit:6 http://linux.teamviewer.com/deb stable InRelease
Hit:7 http://linux.teamviewer.com/deb preview InRelease
Reading package lists... Done
+ sudo apt-get install -y clang-5.0 clang++-5.0
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'clang-5.0-doc' for regex 'clang++-5.0'
Note, selecting 'clang-5.0' for regex 'clang++-5.0'
Note, selecting 'python-clang-5.0' for regex 'clang++-5.0'
Note, selecting 'libclang-5.0-dev' for regex 'clang++-5.0'
Note, selecting 'clang-5.0-examples' for regex 'clang++-5.0'
clang-5.0 is already the newest version (1:5.0-3~16.04.1).
clang-5.0-doc is already the newest version (1:5.0-3~16.04.1).
clang-5.0-examples is already the newest version (1:5.0-3~16.04.1).
libclang-5.0-dev is already the newest version (1:5.0-3~16.04.1).
python-clang-5.0 is already the newest version (1:5.0-3~16.04.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ sudo apt-get install -y unzip
Reading package lists... Done
Building dependency tree
Reading state information... Done
unzip is already the newest version (6.0-20ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ export C_COMPILER=clang-5.0
+ C_COMPILER=clang-5.0
+ export COMPILER=clang++-5.0
+ COMPILER=clang++-5.0
[... 273 lines cut ...]
-- Targeting AArch64
-- Targeting AMDGPU
-- Targeting ARM
-- Targeting BPF
-- Targeting Hexagon
-- Targeting Lanai
-- Targeting Mips
-- Targeting MSP430
-- Targeting NVPTX
-- Targeting PowerPC
-- Targeting Sparc
-- Targeting SystemZ
-- Targeting X86
-- Targeting XCore
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    LIBCXX_INSTALL_EXPERIMENTAL_LIBRARY

-- Build files have been written to: /home/yonhdee/airsim/AirSim/llvm-build
+ make cxx
make: *** No rule to make target 'cxx'.  Stop.
```

yonhdee, 2018-07-01:
It seems to be a network issue, github connectivity isn't stable in my location, the unfinished files cheated the script, which caused file missing error

closed 2018-07-01 by yonhdee

rorypeck, 2018-07-31:
I have met the same errors, don't know how to solve the problem

rorypeck, 2018-07-31:
in setup.sh files, there are the following lines:
```
# Below is alternative way to get cland by downloading binaries
# get clang, libc++
# sudo rm -rf llvm-build
# mkdir -p llvm-build/output
# wget "http://releases.llvm.org/4.0.1/clang+llvm-4.0.1-x86_64-linux-gnu-debian8.tar.xz"
# tar -xf "clang+llvm-4.0.1-x86_64-linux-gnu-debian8.tar.xz" -C llvm-build/output

# #other packages - not need for now
# #sudo apt-get install -y clang-3.9-doc libclang-common-3.9-dev libclang-3.9-dev libclang1-3.9 libclang1-3.9-dbg libllvm-3.9-ocaml-dev libllvm3.9 libllvm3.9-dbg lldb-3.9 llvm-3.9 llvm-3.9-dev llvm-3.9-doc llvm-3.9-examples llvm-3.9-runtime clang-format-3.9 python-clang-3.9 libfuzzer-3.9-dev

#get libc++ source
if [[ ! -d "llvm-source-50" ]]; then 
    git clone --depth=1 -b release_50  https://github.com/llvm-mirror/llvm.git llvm-source-50
    git clone --depth=1 -b release_50  https://github.com/llvm-mirror/libcxx.git llvm-source-50/projects/libcxx
    git clone --depth=1 -b release_50  https://github.com/llvm-mirror/libcxxabi.git llvm-source-50/projects/libcxxabi
else
    echo "folder llvm-source-50 already exists, skipping git clone..."
fi

#build libc++
if [ "$(uname)" == "Darwin" ]; then
    rm -rf llvm-build
else
    sudo rm -rf llvm-build
fi
mkdir -p llvm-build
pushd llvm-build >/dev/null

"$CMAKE" -DCMAKE_C_COMPILER=${C_COMPILER} -DCMAKE_CXX_COMPILER=${COMPILER} \
      -LIBCXX_ENABLE_EXPERIMENTAL_LIBRARY=OFF -DLIBCXX_INSTALL_EXPERIMENTAL_LIBRARY=OFF \
      -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=./output \
            ../llvm-source-50

make cxx
```
**so when meeting the errors, just down the libcxx manually to solve the problem:**
``
    git clone --depth=1 -b release_50  https://github.com/llvm-mirror/libcxx.git llvm-source-50/projects/libcxx
    git clone --depth=1 -b release_50  https://github.com/llvm-mirror/libcxxabi.git llvm-source-50/projects/libcxxabi
```