microsoft/AirSim #1218: Linking problem when separating c++ HelloDrone

kevindehecker, 2018-07-01:
I'm attempting to use airsim from existing code (c++ on Linux). I tried to get the sample projects to work in an external project, but I'm having difficulties. 

I have made a skeleton version of the project here: https://github.com/kevindehecker/airsim-test
The code is an exact copy of the HelloDrone example, the only differences are in CMakeList.txt, which I need to be standalone.
The project compiles, but does not link. I get a whole bunch of the following error:

```
Scanning dependencies of target airsimtest
[ 50%] Building CXX object CMakeFiles/airsimtest.dir/main.cpp.o
[100%] Linking CXX executable airsimtest
/home/houjebek/code/AirSim/build_debug/output/lib/librpc.a(client.cc.o): In function `client':
/home/houjebek/code/AirSim/external/rpclib/rpclib-2.2.1/lib/rpc/client.cc:167: undefined reference to `std::__1::to_string(int)'
/home/houjebek/code/AirSim/external/rpclib/rpclib-2.2.1/lib/rpc/client.cc:167: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
/home/houjebek/code/AirSim/external/rpclib/rpclib-2.2.1/lib/rpc/client.cc:175: undefined reference to `std::__1::thread::~thread()'
/home/houjebek/code/AirSim/external/rpclib/rpclib-2.2.1/lib/rpc/client.cc:167: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
/home/houjebek/code/AirSim/build_debug/output/lib/librpc.a(client.cc.o): In function `thread<(lambda at /home/houjebek/code/AirSim/external/rpclib/rpclib-2.2.1/lib/rpc/client.cc:169:27), void>':
/home/houjebek/code/AirSim/llvm-build/output/include/c++/v1/thread:362: undefined reference to `std::__1::__thread_struct::__thread_struct()'
/home/houjebek/code/AirSim/llvm-build/output/include/c++/v1/thread:372: undefined reference to `std::__1::__throw_system_error(int, char const*)'
/home/houjebek/code/AirSim/build_debug/output/lib/librpc.a(client.cc.o): In function `std::__1::default_delete<std::__1::__thread_struct>::operator()(std::__1::__thread_struct*) const':
/home/houjebek/code/AirSim/llvm-build/output/include/c++/v1/memory:2233: undefined reference to 
`std::__1::__thread_struct::~__thread_struct()'
/home/houjebek/code/AirSim/llvm-build/output/include/c++/v1/memory:2233: undefined reference to `std::__1::__thread_struct::~__thread_struct()'
/home/houjebek/code/AirSim/build_debug/output/lib/librpc.a(client.cc.o): In function `unique_lock':
/home/houjebek/code/AirSim/llvm-build/output/include/c++/v1/__mutex_base:131: undefined reference to `std::__1::mutex::lock()'
...
```

The original HelloDrone project itself in AirSim runs without problem, so it must be something I'm doing wrong in the cmake settings. Where to look?

wardzacharyj, 2018-07-13:
Try adding this to the top of your cmake (assuming you have clang++ 5.0 installed)
 
``cmake
set(CMAKE_CXX_COMPILER /usr/bin/clang++)
``

kevindehecker, 2018-07-14:
Hmm, that would change the compiler for my own project, right? That is not desirable as I would like to keep things at gcc, at least for my own code.

kevindehecker, 2018-07-14:
Good news, in the end I did manage to solve this issue, by including the cmake modules of Airsim directly. See the update skeleton project at: https://github.com/kevindehecker/airsim-test
The only small thing that is nagging me still is that during compilation of the airsim library directly to my own build folder, a lot of warnings are generated. Not sure how to get rid of those yet...

But in any case, you can copy this cmakelist into your own project, set the airsim root, and you can use Airsim from an outside gcc C++ project. 

@ the airsim devs, I think it would be nice if there is a simple example of an external project, like I made here. If the Airsim devs want me to create a PR out of it let me know.

wardzacharyj, 2018-07-18:
@kevindehecker I tried out your cmake but I am running into the following error. Did you encounter this during your testing?

```
home/blah/blah/example_project/AirSim/AirLib/include/common/VectorMath.hpp: 

In static member function ‘static QuaternionT msr::airlib::VectorMathT<Vector3T, QuaternionT, RealT>::lookAt(Vector3T, Vector3T)’:
home /blah/blah/AirSim/AirLib/include/common/VectorMath.hpp:505:21: error: ‘acosf’ is not a member of ‘std’
         RealT ang = std::acosf(dot);
                     ^~~
```

madratman, 2019-02-13:
fwiw, I was doing something similar, but the only way I could get it to compile was by using clang 5.0 and calling the `CommonSetup()` macro by uncommenting Line 14 here https://github.com/kevindehecker/airsim-test/blob/master/CMakeLists.txt#L14

madratman, 2019-04-06:
update on this. I was able to compile a standalone project with gcc 5.4.0, with airsim built with clang 5.0. 
it's available here https://github.com/madratman/airsim_standalone

closed 2019-04-06 by madratman