microsoft/AirSim #1219: Error with update_from_git.bat on Windows 10 - The system cannot find the file specified

ngwaihung, 2018-07-02:
Hi all, I'm currently on Windows 10 (64-bit), using Visual Studio Community 2017. Having tried running **update_from_git.bat** (as per build instructions for blocks environment [here](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md)) as Admin from directory, and via **x64 Native Tools Command Prompt for VS 2017**, I kept encountering the below stated error.
```
Using AirSimPath = D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\

            Newer                 13.1 m        D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\lib\x64\Debug\MavLinkCom.bsc
            Newer                 12.9 m        D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\lib\x64\Release\MavLinkCom.bsc

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>rd /s/q Build
The system cannot find the file specified.

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>rd /s/q Binaries
The system cannot find the file specified.

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>rd /s/q Intermediate
The system cannot find the file specified.

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>rd /s/q Saved
The system cannot find the file specified.

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>rd /s/q Plugins\AirSim\Binaries
The system cannot find the file specified.

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>rd /s/q Plugins\AirSim\Intermediate
The system cannot find the file specified.

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>rd /s/q Plugins\AirSim\Saved
The system cannot find the file specified.

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>del *.sln
Could Not Find D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks\*.sln

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>pause
Press any key to continue . . .
```

This is then followed by...

```
D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>setlocal

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>del /q gen_temp.txt
Could Not Find D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks\gen_temp.txt

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>powershell -command "& { (Get-ItemProperty 'Registry::HKEY_CLASSES_ROOT\Unreal.ProjectFile\shell\rungenproj' -Name 'Icon' ).'Icon' } > gen_temp.tmp"
Get-ItemProperty : Cannot find path 'HKEY_CLASSES_ROOT\Unreal.ProjectFile\shell\rungenproj' because it does not exist.
At line:1 char:6
+ & { (Get-ItemProperty 'Registry::HKEY_CLASSES_ROOT\Unreal.ProjectFile ...
+      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (HKEY_CLASSES_RO...hell\rungenproj:String) [Get-ItemProperty], ItemNotFo
   undException
    + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.GetItemPropertyCommand

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>type gen_temp.tmp  1>gen_temp.txt

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>set /p gen_bin= 0<gen_temp.txt

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>del /q gen_temp.tmp

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>del /q gen_temp.txt

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>for %f in (*.uproject) do (
echo Generating files for %f
 /projectfiles "D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks\%f"
)

D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>(
echo Generating files for Blocks.uproject
 /projectfiles "D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks\Blocks.uproject"
)
Generating files for Blocks.uproject
'/projectfiles' is not recognized as an internal or external command,
operable program or batch file.
Press any key to continue . . .
```

This is my directory listing
```
 Directory of D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks

02/07/2018  01:01 PM    <DIR>          .
02/07/2018  01:01 PM    <DIR>          ..
02/07/2018  10:58 AM               352 .gitignore
02/07/2018  10:58 AM               323 Blocks.uproject
02/07/2018  11:42 AM               193 clean.bat
02/07/2018  10:58 AM               445 clean.sh
02/07/2018  10:58 AM    <DIR>          Config
02/07/2018  10:58 AM    <DIR>          Content
02/07/2018  10:58 AM               391 GenerateProjectFiles.bat
02/07/2018  10:58 AM               608 GenerateProjectFiles.sh
02/07/2018  10:58 AM             1,958 package.bat
02/07/2018  11:17 AM    <DIR>          Plugins
02/07/2018  10:58 AM    <DIR>          Source
02/07/2018  11:35 AM               995 update_from_git.bat
02/07/2018  10:58 AM               381 update_from_git.sh
02/07/2018  10:58 AM               234 update_to_git.bat
02/07/2018  10:58 AM               379 update_to_git.sh
```

Would appreciate any advice on how to proceed with troubleshooting, thanks!

SweetShot, 2018-07-02:
`D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>powershell -command "& { (Get-ItemProperty 'Registry::HKEY_CLASSES_ROOT\Unreal.ProjectFile\shell\rungenproj' -Name 'Icon' ).'Icon' } > gen_temp.tmp"
Get-ItemProperty : Cannot find path 'HKEY_CLASSES_ROOT\Unreal.ProjectFile\shell\rungenproj' because it does not exist.
At line:1 char:6
+ & { (Get-ItemProperty 'Registry::HKEY_CLASSES_ROOT\Unreal.ProjectFile ...
+      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (HKEY_CLASSES_RO...hell\rungenproj:String) [Get-ItemProperty], ItemNotFo
   undException
    + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.GetItemPropertyCommand`

This error is coming cause either you have not installed Unreal Engine 4.18 or havent restarted epic games launcher after installing. After restart it will ask you to associate project file extension with unreal engine, click on fix now to fix it and re-run the update_from_git.bat
That will solve the problem.

ngwaihung, 2018-07-03:
Hi SweetShot,

Thanks for the solution! That fixed most of the error. However to my understanding from https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md there is supposed to be a VS solution file generated in the process, but after executing the batch file from terminal again there are no new additions or modifications to my **%dir%/AirSim/Unreal/Environments/Blocks** directory apart from a new "Intermediate" folder within which there are some files related to build rules but nothing of particular use. This is my log post-restarting my computer and attempting to run the batch file.

```
Using AirSimPath = D:\git\AirSim\

        *EXTRA Dir        -1    D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\
          *EXTRA File                  5        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\.gitignore
          *EXTRA File              23307        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\AirLib.vcxproj
          *EXTRA File              20730        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\AirLib.vcxproj.filters
          *EXTRA File                198        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\update_mavlibkcom.bat
        *EXTRA Dir        -1    D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\
        *EXTRA Dir        -1    D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\api\
          *EXTRA File               2393        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\api\ApiProvider.hpp
          *EXTRA File                441        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\api\ApiServerBase.hpp
          *EXTRA File              14271        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\api\RpcLibAdapatorsBase.hpp
          *EXTRA File               3359        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\api\RpcLibClientBase.hpp
          *EXTRA File               2122        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\api\RpcLibServerBase.hpp
          *EXTRA File               4106        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\api\VehicleApiBase.hpp
          *EXTRA File               3047        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\api\VehicleSimApiBase.hpp
          *EXTRA File               1061        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\api\WorldSimApiBase.hpp
        *EXTRA Dir        -1    D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\
          *EXTRA File              48541        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\AirSimSettings.hpp
          *EXTRA File               2266        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\CancelToken.hpp
          *EXTRA File               2490        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\ClockBase.hpp
          *EXTRA File                936        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\ClockFactory.hpp
          *EXTRA File               2308        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\Common.hpp
          *EXTRA File               7600        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\CommonStructs.hpp
          *EXTRA File               1945        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\DelayLine.hpp
          *EXTRA File               7458        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\EarthCelestial.hpp
          *EXTRA File              18499        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\EarthUtils.hpp
          *EXTRA File               2411        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\FirstOrderFilter.hpp
          *EXTRA File               3706        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\FrequencyLimiter.hpp
          *EXTRA File               2298        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\GaussianMarkov.hpp
          *EXTRA File               8185        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\GeodeticConverter.hpp
          *EXTRA File               2031        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\ImageCaptureBase.hpp
          *EXTRA File               1681        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\LogFileWriter.hpp
          *EXTRA File               2151        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\PidController.hpp
          *EXTRA File               2574        D:\git\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\ScalableClock.hpp
[... cut ...]
02/07/2018  03:48 PM               381 update_from_git.sh
02/07/2018  03:48 PM               234 update_to_git.bat
02/07/2018  03:48 PM               379 update_to_git.sh
```

Is the solution file supposed to be generated in the same folder as where the batch file is ran? If so there doesn't seem to be any visible output from running the batch file. Perhaps to note in case it's of any use to troubleshooting is that I ran the batch file from x64 Native Tools Command Prompt too as executing it as a batch file by itself redirects me to my OS' drive.

closed 2018-07-03 by ngwaihung

ArtemBoyarintsev, 2019-07-04:
I had a problem that *update_from_git.bat* didn't generate solution for Visual studio. However, **Update 2 from 3rd July 2018** didn't help me. I would like to say what helped me. 

**My solution:**
On the AirSim install page it is told to ensure that Windows 8.1 SDK will be installed. So **exactly Windows 8.1 SDK is required** (while I was thinking that Windows 10 SDK would do, when it wouldn't). 

**Hint:**
If my solution doesn't help you, there are logs in _saved/logs/_, try to see there :)

MaxwellHuang2015, 2019-09-22:
A little bit addition to SweetShot's solution~ Be attention to quit "Epic Game Launcher" after installation of Unreal Engine 4.18, and restart "Epic Game Launcher" rather than start "Unreal Engine" which was newly added on the desktop. The fix of associate project file extension with unreal engine will only start here, after which the issue will be fixed

abhigoku10, 2019-10-01:
@ArtemBoyarintsev @MaxwellHuang2015  i am also facing the same issue pointed out by @ngwaihung  was able to solve the first issues by fix issues with the epic launcher  but there is not .sln files created by durign the build , is it necessary to have only the windows 8.1 version , currently i am having 10

ArtemBoyarintsev, 2019-10-01:
@abhigoku10 Well, I was able to run it on windows 10. I don't exactly remember what actions I made, but the main point is to install Windows 8.1 SDK. This sdk can be installed on windows 10 as well. I also remember, I have deleted Windows 10 SDK before installing 8.1 SDK.

caesar84, 2020-02-26:
I am having the same problem and its [https://github.com/microsoft/AirSim/issues/2432](here)
any suggestions?

kritikagovindaraj, 2020-06-02:
https://github.com/Microsoft/AirSim/issues/1219#issuecomment-533849033
This solution worked.

rakehsaleem, 2021-06-03:
> `D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>powershell -command "& { (Get-ItemProperty [...]

This one works for me and I had the same error at first, but following this my .sln files were build :)

zclyaya, 2022-11-28:
[image]
how to solve this problem?

MaxwellHuang2015, 2022-11-28:
这是来自QQ邮箱的假期自动回复邮件。你好，我最近正在休假中，无法亲自回复你的邮件。我将在假期结束后，尽快给你回复。