microsoft/AirSim #1220: Issue with Python Client, vehicle name = ''

SweetShot, 2018-07-02:
Hello,
    I am getting following error while trying to execute hello drone script from python client
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for '' is not available. This could either because this is simulation-only API or this vehicle does not exist.

I do understand that this must be some small problem of setting vehicle_name param but I have no idea what to set it to.
Also I have tried adding Vehicle in Json with key "Drone" and using Drone as vehicle_name but doesnt work
 `  "Vehicles": {
    "Drone": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": true
      }
    }
  }`

Thank you

ngwaihung, 2018-07-03:
Facing the same issue here as SweetShot pointed out.
Something worth noting perhaps is that, vehicle_name from within the simulation seems to be blank by default (i.e. "").
I'm wondering if there are any other settings for the simulation that needs to be configured because as it is currently, it seems that **Vehicle Sim-API** works while **Vehicle API** does not.

This can be verified by calling simGetVehiclePose - it works fine as shown below.
```
<Pose> {   'orientation': <Quaternionr> {   'w_val': 0.942059338092804,
    'x_val': -0.0,
    'y_val': 0.0,
    'z_val': 0.33544620871543884},
    'position': <Vector3r> {   'x_val': 0.9092577695846558,
    'y_val': 0.7414843440055847,
    'z_val': 56.27068328857422}}
```
And if you set a vehicle name in the parameter that is not "", it will return an appropriate error message
`rpclib: function 'simGetVehiclePose' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle Sim-API for 'asdf' is not available. This could either because this is not a simulation or this vehicle does not exist.`

In other words, the functions with a 'sim' prefix (simGetVehiclePose, simGetCameraInfo, etc) works fine, but not the rest.

mitairlines, 2018-07-04:
I have the same problem

sytelus, 2018-07-07:
Thanks for reporting this issue. This is now fixed in latest master.

closed 2018-07-07 by sytelus

trungvm, 2021-04-06:
that error because airsim didn't connected with simulation SITL or HITL

mingming81, 2024-03-19:
Remove the argument concerning the drone_name works for me. It is a way used by DRLwithTL project presented in https://towardsdatascience.com/deep-reinforcement-learning-for-drones-in-3d-realistic-environments-36821b6ee077