microsoft/AirSim #1224: how to run airsim in headless mode

husha1993, 2018-07-04:
https://github.com/Microsoft/AutonomousDrivingCookbook/issues/34

Are there anyway I can just run my model with AirSim on a command line in Linux or windows terminal except use the binaries version mentioned in the link above?

SweetShot, 2018-07-04:
Not sure about that but have a look ar NoDisplay setting

madratman, 2019-04-10:
set this environment variables:
```
export SDL_VIDEODRIVER_VALUE=offscreen
export SDL_HINT_CUDA_DEVICE=0
```
If you plan to use docker, we have an headless option there as well. please see this page https://github.com/Microsoft/AirSim/blob/master/docs/docker_ubuntu.md

closed 2019-04-10 by madratman

mchowdhury-tft, 2019-08-29:
where do you set this "export SDL_VIDEODRIVER_VALUE=offscreen
export SDL_HINT_CUDA_DEVICE=0" variable for the headless mode ? 
Please let me know....

ivannson, 2019-12-24:
@mchowdhury-tft just paste each command in terminal and run it

Neabfi, 2020-01-08:
I’ve followed those steps but it is stuck on:
`[2020.01.08-01.13.53:613][  3]LogRenderer: Reallocating scene render targets to support 128x128 Format 10 NumSamples 1 (Frame:0).`
Any ideas?

Neabfi, 2020-01-16:
@mchowdhury-tft  Did you manage to make it work?

For me it just remains frozen forever after the line:
`LogRenderer: Reallocating scene render targets to support 128x128 Format 10 NumSamples 1 (Frame:0).`

jeffreyng99, 2020-08-24:
Hello @madratman and @Neabfi I am running the docker container headless, however I am running into the same problem as you said. Whenever I run AirSim not in headless mode, I have to click the play button to start the simulator and rpc server. How would start the simulator in headless mode?

thomasTUK, 2021-06-04:
Also wondering the same. How to do it?