microsoft/AirSim #1225: msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception.

koukoulala, 2018-07-04:
**error:**
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for '' is not available. This could either because this is simulation-only API or this vehicle does not exist.

my python --version=3.6;   system: mac os;    UE:4.18；    Environment:Blocks
I want to use python API.so write "python hello_car.py".Although conneted,but got this error.I really don't know what to do.
**ps**:Here is my json file:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
  "SettingsVersion": 1.2,
  "SimMode":"Car",
  "Vehicles":{
    "PhysXCar": {
        "VehicleType": "PhysXCar",
        "AllowAPIAlways": true,
        "AutoCreate": true,
        "RC": {
            "RemoteControlID": 0,
            "AllowAPIWhenDisconnected":true
        }
    }
  }
}

SweetShot, 2018-07-04:
Copy of [#1220](https://github.com/Microsoft/AirSim/issues/1220)

sytelus, 2018-07-07:
Thanks for reporting this issue. This is now fixed in latest master.

closed 2018-07-07 by sytelus

Josepharol, 2020-09-19:
@sytelus I get this same error when I try to add more than two cars in the multi_agent_car.py in the city environment.

Error message:

> (base) D:\Airsim\PythonClient\car>python multi_agent_car.py
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "multi_agent_car.py", line 39, in <module>
    client.enableApiControl(True, "Car3")
  File "C:\ProgramData\Anaconda3\lib\site-packages\airsim\client.py", line 67, in enableApiControl
    self.client.call("enableApiControl", is_enabled, vehicle_name)
  File "C:\ProgramData\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\ProgramData\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'Car3' is not available. This could either because this is simulation-only API or this vehicle does not exist.

Please find the code for multi agent - I run it in anaconda power shell after opening the city environment run file.
`import airsim
import cv2
import numpy as np
import os
import setup_path 
import time

# Use below in settings.json with blocks environment
"""
{
	"SettingsVersion": 1.2,
	"SimMode": "Car",
	
	"Vehicles": {
		"Car1": {
		  "VehicleType": "PhysXCar",
		  "X": 4, "Y": 0, "Z": -2
		},
		"Car2": {
		  "VehicleType": "PhysXCar",
		  "X": -4, "Y": 0, "Z": -2
		},
        "Car3": {
        "VehicleType": "PhysXCar",
        "X": -50,
        "Y": -31,
        "Z": -2,
        "Yaw": 180
        }
    }
}
"""

# connect to the AirSim simulator 
client = airsim.CarClient()
client.confirmConnection()
client.enableApiControl(True, "Car1")
client.enableApiControl(True, "Car2")
client.enableApiControl(True, "Car3")

car_controls1 = airsim.CarControls()
car_controls2 = airsim.CarControls()
car_controls3 = airsim.CarControls()

for idx in range(3):
    # get state of the car
    car_state1 = client.getCarState("Car1")
    print("Car1: Speed %d, Gear %d" % (car_state1.speed, car_state1.gear))
    car_state2 = client.getCarState("Car2")
    print("Car1: Speed %d, Gear %d" % (car_state2.speed, car_state2.gear))
[... 82 lines cut ...]

    for response in responses1 + responses2 + responses3:
        filename = 'c:/temp/car_multi_py' + str(idx)

        if response.pixels_as_float:
            print("Type %d, size %d" % (response.image_type, len(response.image_data_float)))
            airsim.write_pfm(os.path.normpath(filename + '.pfm'), airsim.get_pfm_array(response))
        elif response.compress: #png format
            print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
            airsim.write_file(os.path.normpath(filename + '.png'), response.image_data_uint8)
        else: #uncompressed array
            print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
            img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8) # get numpy array
            img_rgb = img1d.reshape(response.height, response.width, 3) # reshape array to 3 channel image array H X W X 3
            cv2.imwrite(os.path.normpath(filename + '.png'), img_rgb) # write to png

#restore to original state
client.reset()

client.enableApiControl(False)

            
`

Is there anything wrong with my code?