microsoft/AirSim #1226: No GPS lock

hadroncolided, 2018-07-04:
Hi everyone.
I am using UE 4.18.3 with Air Sim and pixhawk v1.8.0 for HIL simulation and I am unable to get GPS lock in QGroundControl. QGroundControl shows 10 satellites and points Redmond (which is how it is supposed to work as far as i now). Also interesting is that when I fly around the UE environment the point in QGroundControl map is also moving, until it suddenly goes back to home position.

I hope that someone can help!

chmi0611, 2018-07-13:
Hi, can you upload your settings.json? I can only fly with PX4 v1.4.4. PX4 v1.8.0 does not respond in HIL.

I'm sorry I cannot help you.

clovett, 2018-12-23:
Recent fixes + pull request [1654](https://github.com/Microsoft/AirSim/pull/1654) make PX4 v1.8.2 work nicely with AirSim.  Please try these bits, thanks.  There is also a pending [declination bug](https://github.com/Microsoft/AirSim/issues/1340) which you can work around.

closed 2019-10-03 by msb336