microsoft/AirSim #1227: Python Client not connecting to AirSim

danielcomas25, 2018-07-04:
Hello, 

I have been trying to install AirSim, I manage to run the Block Environment that is in the examples. However, when I try to run the Python Client, there is always **an error with the communication connection**. As follows:

WARNING:tornado.general:Connect error on fd 352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 352: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 660: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 660: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 660: WSAECONNREFUSED
Traceback (most recent call last):
  File "hello_car.py", line 10, in <module>
    client.confirmConnection()
  File "C:\Users\Daniel\Documents\Visual Studio 2017\Air\AirSim\PythonClient\airsim\client.py", line 54, in confirmConnection
    if self.ping():
  File "C:\Users\Daniel\Documents\Visual Studio 2017\Air\AirSim\PythonClient\airsim\client.py", line 24, in ping
    return self.client.call('ping')
  File "C:\Users\Daniel\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\Daniel\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 43, in get
    raise self._error
**msgpackrpc.error.TransportError: Retry connection over the limit**

I have **disabled the firewall and the antivirus** like I read in other issues but I still do not manage to connect them. **While trying to connect I am running the Block Environment and AirSim** 

I am running the Block Enviroment, then AirSim.sln and after the PythonClient, hello_car.py. AirSim.sln , when running, it is always displaying this:

==||=>
        Welcome to DroneShell 1.0.
        Type ? for help.
        Microsoft Research (c) 2017.

Waiting for connection - XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

koukoulala, 2018-07-05:
did you open the simulator on the UE4?first run airsim,then python hello_car.py.

danielcomas25, 2018-07-05:
Yes, I try that and still the client can't reach. Airsim is still showing **Waiting for connection** and Python Client is showing **WARNING:tornado.general:Connect error on fd 352: WSAECONNREFUSED**

danielcomas25, 2018-07-06:
For anyone having the same problem. I recommend you to download the version AirSim 1.2. Then, follow this tutorial for running AirSim from Unreal. https://www.youtube.com/watch?v=1oY8Qu5maQQ&t=324s  After that, you should be able to run the python client without any problem using Anaconda 4.4

closed 2018-07-06 by danielcomas25

ArthurYdalgo, 2021-07-10:
I replied to a [similar issue](https://github.com/microsoft/AirSim/issues/1949) [what could be the solution](https://github.com/microsoft/AirSim/issues/1949#issuecomment-877612274)