microsoft/AirSim #1232: In v1.2, when model was set to hexacopter, simulation crashed on start

adamkeynes, 2018-07-06:
i start ue4 and my project with vs, when click PLAY in ue4, it says something is wrong with UE4Editor-AirSim-Win64-DebugGame.dll and break.

i cannot find any obvious difference in Px4MultiRotorParams.hpp,  guess something maybe wrong with initializeRotorHexX()

adamkeynes, 2018-07-07:
[image]
this is exactly what happened

adamkeynes, 2018-07-07:
[image]

KeyneZzz, 2018-07-30:
@sytelus can you pls check if there is any problem with hexa? and by the way check #1234  pls, thx

linarism, 2018-08-17:
@KeyneZzz I also ran into the same issue. The iterator tries to update `rotor_speed` for all propellers, but `rotating_movements_` is only allocated for a hardcoded 4 propellers defined in `FlyingPawn.h`. Increasing the `rotor_count` value in that file will solve your problem.

KeyneZzz, 2018-08-17:
@linarism thx a lot

lovettchris, 2019-04-16:
Here's a hexacopter that works: https://github.com/Microsoft/AirSim/pull/1890

closed 2020-02-14 by ironclownfish