microsoft/AirSim #1237: Python Api still not working after update

vaschindler, 2018-07-09:
Hey,

I got the newest build of AirSim with the Python Api fix and still have problems with the settings file. In my settings.json I just have 

```
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor"
}

```
When I press play to simulate, I got the following error message: 

`"There were no compatible vehicles created for current SimMode! Check your settings.json."`

But the drone got spawned after all. Now I type in python hello_drone.py and get: 
```

msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. 
The exception contained this information: Vehicle API for '' is not available. 
This could either because this is simulation-only API or this vehicle does not exist
```

Is there something I've missed? I had no problems with the API control before the new API changes.

Beijingsheng, 2018-08-09:
Same problem here. Could anyone help?

AloshkaD, 2018-09-04:
I've had this issue after I accidentally clicked on some button or option inside the unreal simulator. Luckily I had my env backed up and everything worked after I restored it. It's an empty env and feel free to give it a try if you need it https://www.dropbox.com/s/ia0qii81k4lyybz/Blocks-minimized.zip?dl=0

AloshkaD, 2018-09-04:
and here is my setting json btw

`{
	"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
	"SettingsVersion": 1.2,
	"SimMode": "Multirotor",
	"ClockSpeed": 1,
	 
	
	"Vehicles": {
		"Drone1": {
		  "VehicleType": "SimpleFlight",
		  "X": 26, "Y": -1  , "Z": -2
 
		},
		"Target1": {
		  "VehicleType": "SimpleFlight",
		  "X": 34, "Y": -4  , "Z": -2
 
		}
    }
}`

msb336, 2019-10-07:
Closing this thread due to inactivity. If you are still struggling with this issue please re-open the thread.

closed 2019-10-07 by msb336