microsoft/AirSim #1239: settings file deleted at startup

rhalaly, 2018-07-09:
I can see that the settings.json file is deleted when the simulator starts. 
No matter what I write there, it ignores it and overwrites my file with the default file template:

> {
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0
}

nathantsoi, 2018-07-14:
same here, any suggestions?

allen9273, 2018-07-27:
Because your folder path including Asian characters.

closed 2018-07-28 by rhalaly

Beijingsheng, 2018-08-08:
I changed the folder path from C:\Users\日升\Documents\AirSim    to   C:\Users\risheng\Documents\AirSim, still settings_json.json will delete what I added.

allen9273, 2018-08-08:
参考 #1066，it works for me

Beijingsheng, 2018-08-09:
I just found that this problem only occurs to settings.json created by CityEnvironment, I can change settings.json created by AirsimNH and it won't be rewritten. Could anyone help with this?

Beijingsheng, 2018-08-09:
Problem solved.