microsoft/AirSim #1244: How to control other cars driving behaviors in Airsim

BrianN92, 2018-07-12:
I have tried the latest version of City map in Airsim. I am wondering how do you guys control the behaviors of other cars besides SUV agent. Those cars seem like they are doing like "spline follow", but they also have the ability to wait for the traffic lights, which looks so cool. Could anybody tell me how to set up other vehicle agents like that? Thank you!!

JordanWang0223, 2018-07-17:
hello, I wonder ask you a question, how do u add another cars in the same maps?

BrianN92, 2018-07-17:
@JordanWang0223 What I did was to create a self-built environment in UE4 and then build the AirSim with it. So I could add as many agents as I need. However, the SUV is the only agent controlled by Airsim. Therefore, I am wondering how to set up other cars like what they did in the "City map".

Beijingsheng, 2018-08-09:
I'm also curious about this. Seems like those cars are pretty smart, I wonder if I can find their controlling code and edit them for my own use.

Misslindalian, 2019-02-23:
Hello.Has the problem been solved? I also want to do this. Thanka a lot!

msb336, 2019-10-07:
Cars are indeed on a spline follow in the City map. The intersection logic employed, unfortunately, is custom built for the environment and is not open source, so we are unable to provide the toolbox we used to construct this feature.

closed 2019-10-07 by msb336