microsoft/AirSim #1246: data mismatch between simGetVehiclePose() and simSetVehiclePose()

husha1993, 2018-07-14:
>>> stepsize= [0,0,10]
>>> airsim_client.simSetVehiclePose(airsim.Pose(airsim.Vector3r(currentposition.x_val+ stepsize[0], currentposition.y_val+stepsize[1], currentposition.z_val+stepsize[2]), airsim.to_quaternion(0, 0, 20)), True)
>>> currentposition.x_val+ stepsize[0]
3.0517577442878974e-07
>>> currentposition.y_val+stepsize[1]
0.0
>>> currentposition.z_val+stepsize[2]
**21.820711135864258**
>>> currentposition = airsim_client.simGetVehiclePose().position
>>> print(currentposition)
<Vector3r> {   'x_val': 3.0517577442878974e-07,
    'y_val': 0.0,
    'z_val': **11.820711135864258**}

thias15, 2018-07-16:
Looks like you add 10 in the line "currentposition.z_val+stepsize[2]" in which case there is no mismatch.

husha1993, 2018-07-17:
stepsize= [0,0,10]
airsim_client.simSetVehiclePose(airsim.Pose(airsim.Vector3r(currentposition.x_val+ stepsize[0], currentposition.y_val+stepsize[1], currentposition.z_val+stepsize[2]), airsim.to_quaternion(0, 0, 20)), True)
currentposition.x_val+ stepsize[0]
3.0517577442878974e-07
currentposition.y_val+stepsize[1]
0.0
currentposition.z_val+stepsize[2]
21.820711135864258
#currentposition.z_val+stepsize[2]
21.820711135864258 is what i set the new position by simSetVehiclePose
currentposition = airsim_client.simGetVehiclePose().position
print(currentposition)
{ 'x_val': 3.0517577442878974e-07,
'y_val': 0.0,
'z_val': 11.820711135864258}
# 'z_val': 11.820711135864258 is the new current position obtained by simGetVehiclePose() after i set the new position 

thus there is mismatch between simSetVehiclePose() and  simGetVehiclePose()...

husha1993, 2018-07-17:
stepsize= [0,0,10]
airsim_client.simSetVehiclePose(airsim.Pose(airsim.Vector3r(currentposition.x_val+ stepsize[0], currentposition.y_val+stepsize[1], currentposition.z_val+stepsize[2]), airsim.to_quaternion(0, 0, 20)), True)
currentposition.x_val+ stepsize[0]
3.0517577442878974e-07
currentposition.y_val+stepsize[1]
0.0
currentposition.z_val+stepsize[2]
21.820711135864258
#currentposition.z_val+stepsize[2]
21.820711135864258 is what i set the new position by simSetVehiclePose
currentposition = airsim_client.simGetVehiclePose().position
print(currentposition)
{ 'x_val': 3.0517577442878974e-07,
'y_val': 0.0,
'z_val': 11.820711135864258}

if there is no mismatch, currentposition = airsim_client.simGetVehiclePose().position
print(currentposition)
should be 
{ 'x_val': 3.0517577442878974e-07,
'y_val': 0.0,
'z_val': 21.820711135864258}
rather than 
print(currentposition)
{ 'x_val': 3.0517577442878974e-07,
'y_val': 0.0,
'z_val': 11.820711135864258}

Gotfeurdom, 2018-07-23:
Try using -10 instead of 10
A value of 10 means your vehicle will be 10m below its current position, which might put it back to its original position if there was a collision or something

husha1993, 2018-07-24:
Hi,

It seems that setVehiclePose only works in computer vision mode, maybe you can check if you have set to 'computer vision mode' in setting.json file.
refer to: https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#setting-pose-in-computer-vision-mode

Gotfeurdom, 2018-07-24:
My mode is set to default (I have to pick between car and copter when I launch the simulation)
I'm baffled by the lack of documentation ... nowhere does it say simSetVehiclePose only works in CV mode nor how it actually works.

Where can I find its implementation in the source code ?
I tried to browse it a bit but I'm lost as there is no comment blocs explaining each files

husha1993, 2018-07-24:
May I know why you use simSetVehiclePose?
If you want to move the car or copter and simulate the physics of the vehicle, you may try the move* api. refer to:
https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#apis-for-multirotor  
If you want to move the camera but don't want to simulate the physics of the vehicle, you can set the mode to computer vision and use the setVehiclePose api. Also, you can try https://unrealcv.org/. 

I didn't try to browse the source code, so I don't know how the move* or set* is implemented.

I'm using airsim, I think it's currently the simulator who can both integrate diverse visual environments and do complex physical simulation.
For other simulators, they are either limited by the number of visual environments or lack of real physical simulation.

Gotfeurdom, 2018-07-24:
Because I don't want to wait forever to put my drone in a specific position ? I'm trying to use AirSim for a reinforcement learning project (which I believe AirSim was built for) ... not fly around for fun, so I need some specific functionality to automatize my learning

Also I use coordinates to navigate on my heightmap and I'm forced to put my playerstart position at (0,0,0) in unreal, because any starting position in AirSim is considered to be (0,0,0) and therefore there is only one starting position that gives me the true coordinates.

I truly liked how AirSim was presented on some youtube videos and I really thought it would be a walk in the park to use it effectively, but the lack of documentation and help I'm getting is driving me crazy

Gotfeurdom, 2018-07-24:
You don't understand

Wherever I place my starting position in unreal, the starting position in AirSim (from simGetVehiclePose for example) will always be (0,0,0) at the start of the simulation and future calls on simGetVehiclePose will always get a set of coordinates relative to that starting position, which is only true is my starting position in Unreal is (0,0,0) as well.
Therefore there is an offset on simGetVehiclePose that I don't have access to (at least using AirSim API and I don't know any way to access it as I'm totally new to unreal)

Gotfeurdom, 2018-07-25:
"Thus you can access to the offset as long as you know your start position in unreal"
I'm not using a simulator/a computer to have to manually set some constants every time I want to change the scene ...
Is there a variable I can access using only my python code ?

Anyway this isn't even a big problem, as I need that "teleport" function anyway and I could easily use it to set my offset right

husha1993, 2018-07-25:
"I'm not using a simulator/a computer to have to manually set some constants every time I want to change the scene ..."
"Is there a variable I can access using only my python code to get the true value of the position in unreal and if so, how ?"

The simulator shouldn't take the responsibility for constructing the scenes, I think this design of airsim is reasonable. But it should take the job for wrapping and reading the scenes constructed by the users who want to test the RL algorithms in the scenes they constructed.

Thus, the offset is something that you already know before run airsim when you construct you scenes.
And when you run the agent of airsim, read the offset, and tell the agent its start position in unreal. That is "reading the scenes".

Gotfeurdom, 2018-07-25:
We can argue all we want, this still doesn't solve my problem ...

husha1993, 2018-07-25:
"Because I don't want to wait forever to put my drone in a specific position ?"
Have you tried the move* api?
eg: client.moveToPositionAsync(-10, 10, -10, 5).join() ?

I'm trying to understand what you want to do with airsim, and according to my understanding, airsim can do what you want.

Gotfeurdom, 2018-07-25:
It's pretty simple.
I want my vehicle to INSTANTLY get to the position I want. (and it would be even better if I can set it to the STATE I want (position, orientation and speed).

The usage is pretty irrelevant but there it is :
I want to build a reinforcement learning algorithm to control the vehicle using client.moveByAngleThrottleAsync(pitch, roll, throttle, yaw_rate, duration)
For my algorithm to generalize my solution, I need to put it in random starting positions, or replay from some known positions

husha1993, 2018-07-26:
"replay from some known positions"

Can you move the agent to some known positions with airsim move and pretend the drone restarts from this position, rather than really disconnet the airsim client and restart the game?

husha1993, 2018-07-26:
Learning to control the vehicle in **low level action** spaces with moveByAngleThrottleAsync and put the vehicle **instantly** to a given state, I feel the two demands are kind of contradictory.

Have you find any other simulator that can satisfy your project?

Gotfeurdom, 2018-07-26:
How are these contradictory ? These two functions would be for totally different purpose

Why would I need to wait to get to point B from point A if I only need to learn something from the surrounding of point B ?

Just because you can pick your nose with your fingers doesn't mean you can't use them for an other purpose

husha1993, 2018-07-26:
"learn something from the surrounding of point B ",
in this case, the agent in point B is controlled in low level;
'get to point B from point A instantly', then the agent from A to B is controlled in high level.
The above is what I referred as contradictory.

Actually, what your project want the airsim to do is to switch between "car"/"multirotor" mode and "computer vision" mode after click the playing button;
While airsim currently seems cannot handle the mode switching during game play: 
https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#simmode

Hoshea7, 2019-07-20:
Hi  @Gotfeurdom, 

have you find any solutions to reset the position after each episode learning?

Gotfeurdom, 2019-07-22:
> 
> 
> Hi @Gotfeurdom,
> 
> have you find any solutions to reset the position after each episode learning?

Hi
No since I've abandoned that simulator since I had the issue.
Also I'm not sure but I think I read somewhere that they finally implemented it

dimikout3, 2019-09-26:
I have also noticed the same problem in using moveToPositionAsync() and simGetVehiclePose.position. Initially I am sending a multicopter  to position (10, 0, -10) and then I retrieve the position using the simGetVehiclePose.position values (for x,y,z coordinates) the result is (x=15.57, y=15.55, z=-9.99). I have also noticed the same problem in other positions. I think that this is a huge bug ... Am I missing anything ?

dimikout3, 2019-09-26:
Same error occurs if I use client.getMultirotorState().kinematics_estimated.position. This make sence since both getMultirotorState() and simGetVehiclePose() use the same function in  airsim/types.py

msb336, 2019-10-03:
@dimikout3
 `moveToPositionAsync()` is somewhat unreliable, as the controller converges upon some non-zero steady-state error. The coordinates that you are getting from `getMultirotorState` and `simGetVehiclePose` are ground truth, and the difference between your input goal position and your ground truth state represents the steady-state error in SimpleFlight's controller.

rajat2004, 2020-04-21:
Came across this issue while adding docstrings for all APIs
`simSetVehiclePose()` seems to work when using Multirotor also. See [teleport.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/teleport.py) for example

PR #3363 (2021-02-02): Fix losing physics after setting pose

Fixes: #1272
Fixes: #1856
Fixes: #1246
Fixes: #3049

## About
This is a slightly modified version of @songxiaocheng's PR originally at #2324. This PR includes all his changes with the exception of 1 line that still has an open question.

## How Has This Been Tested?
Build was confirmed to work, and a modified version of the teleport.py was tested where a moveToPositionAsync() API call was made after a simSetVehiclePose() call.

## Screenshots (if appropriate):
closed 2021-02-02 by zimmy87