microsoft/AirSim #1256: How to load pre-compiled environments in Unreal Editor on Windows

cseas, 2018-07-18:
I've built AirSim on Windows 10 as explained in the [instructions](https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md), and ran the Blocks environment.

I want to load the pre-compiled environments like the Neighbourhood and City in Unreal Editor so I can scale their graphics down and compile them myself so they can perform well on my low-end GPU (Radeon R5 M335).

Is there any way to do this?

Alternatively, can I run the pre-compiled binaries provided in the [latest release](https://github.com/Microsoft/AirSim/releases) in low-graphics mode similar to how I can scale the Blocks environment in Unreal Editor to run on "low" graphics setting?

cseas, 2018-07-19:
Found a work around for this and sent a pull request #1263 

The binaries give a better performance by running in low-resolution mode.
Still curious as to whether we can compile the binaries ourselves and scale the graphics down in Unreal Editor.

eds89, 2019-05-09:
Did you manage to run the pre-compiled versions in Unreal Editor? I want to generate my own version on top of theirs.

cseas, 2019-05-09:
I wasn't able to load the pre-compiled binaries into Unreal Editor. However, I just wanted to run them in low resolution, so my Pull Request above addressed that without any modification to the binaries. Perhaps there are some attributes we can control using the command line, but can't build on top of a binary unless they provide us the source Unreal project for compiling the binary ourselves.

jonyMarino, 2020-08-19:
The enviroments are closed source.

closed 2020-08-19 by jonyMarino