microsoft/AirSim #1257: build project problem

alitarfa, 2018-07-18:
Hi every one i have probelm when i make the build of the project 
the problem related to rpclib @
[image]

Besjan, 2018-07-24:
Same problem here; using latest cmake v3.12.0

alitarfa, 2018-07-25:
i have used the lastest version but the problem is when downloading the rpclib from the github some problem happen

lipanpeng, 2018-08-16:
@alitarfa I also met this error. Can you tell me how you solved it? Thank you!

alitarfa, 2018-08-16:
in this case i have found auth problem during the compilation phase , that is mean when you run the file it will use the internet connection to get rpclib from github and this need some authentication steps i don't know why .
what i have done was to build the cmd file by my hand and to do this first download the rpclib library you will find the link inside the the cmd file just open it and you will find the libraries and follow the instruction inside the cmd file

closed 2020-02-14 by ironclownfish