microsoft/AirSim #1259: Cannot compile Blocks project on Ubuntu

martijn-vinotion, 2018-07-19:
Hello everyone,

I am trying to open the Blocks project on Ubuntu 16.04 but am running into difficulties.

After compiling AirSim, I try to open the Blocks project in the Unreal Editor.
First, it gives the following message

> This project was made with a different version of the Unreal Engine. Converting to this version will rebuild your code projects.
> 
> New features and improvements sometimes cause API changes, which may require you to modify your code before it compiles. Content saved with newer versions of the editor will not open in older versions.
>
> We recommend you open a copy of your project to avoid damaging the original.

After selecting 'open a copy', it first creates a copy, generates project files and then tries to recompile the project and gives the following error:

```
The project could not be compiled. Would you like to open it in Null Source Code Access?

Running ~/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Blocks Development Linux -project="~/AirSim/AirSim/Unreal/Environments/Blocks 4.19 - 3/Blocks.uproject" -editorrecompile -progress -NoHotReloadFromIDE
Scale at X=65.000000, Y=24.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
Scale at X=960.000000, Y=600.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
Scale at X=960.000000, Y=600.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
Setting swap interval to 'Immediate'

Running Mono...

~/UnrealEngine/Engine ~/UnrealEngine/Engine/Binaries/Linux
Using 'git status' to determine working set for adaptive non-unity build.
Creating makefile for Blocks (no existing makefile)
Performing full C++ include scan (no include cache file)
@progress push 5%
Building UnrealHeaderTool...
Using 'git status' to determine working set for adaptive non-unity build.
Creating makefile for UnrealHeaderTool (no existing makefile)
------- Build details --------
Using clang (/usr/bin/clang++) version '5.0.0' (string), 5 (major), 0 (minor), 0 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /usr/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
Performing 1 actions (8 in parallel)
[1/1] Link (lld) UnrealHeaderTool
Total build time: 1.85 seconds (Local executor: 0.11 seconds)
Error: UnrealHeaderTool failed for target 'BlocksEditor' (platform: Linux, module info: ~/AirSim/AirSim/Unreal/Environments/Blocks 4.19 - 3/Intermediate/Build/Linux/B4D820EA/BlocksEditor/Development/BlocksEditor.uhtmanifest, exit code: OtherCompilationError (5)).
```

I then choose to open it with Null Code Access, but then nothing happens.
When I then try to open the created copy, it asks to rebuild libraries (blocks and the airsim plugin), when I choose to do so it fails because blocks could not be compiled.
It then suggests to build manually.
I figured out how to build the airsim plugin manually but I cannot find a way to build the project manually on Ubuntu.

Can anyone help with this?

Mayankm96, 2018-07-19:
As per the error message you have got, are you using Unreal Engine 4.18?

martijn-vinotion, 2018-07-19:
I am using Unreal Engine 4.19.2

Mayankm96, 2018-07-19:
Currently, the plugin is supported only on Unreal Engine 4.18. Check the documentation on the same [here](https://github.com/Microsoft/AirSim/blob/master/docs/build_linux.md#install-and-build).

martijn-vinotion, 2018-07-20:
@Mayankm96 thanks for your reply. I didn't notice that and it might indeed be a good idea to degrade to 4.18.
Coincidentally, I just managed to get the project compiled:

On Ubuntu, the project binaries can be compiled in QT Creator (I followed [this tutorial video](https://www.youtube.com/watch?v=RnINgMbJx5Q)).

Initially Clang would complain that it didn't recognize `-fuse-ld=lld`.
Apparantly, the link to this linker is hardcoded in the UnrealHeaderTool (see source file LinuxToolchain.cs in `UnrealEngine/Engine/Source/Programs/UnrealBuildTool/Platform/Linux/`).

For me, changing the link to this linker to the location where the ld-linker is installed, solved the problem.
I changed `-fuse-ld=lld` to `-fuse-lld=/usr/bin/ld` and recompiled the UnrealHeaderTool by running `make UnrealHeaderTool` in the UnrealEngine folder.

Then I could compile the game in QT Creator and everything worked as it should.

closed 2018-07-20 by martijn-vinotion