microsoft/AirSim #1265: SimSetObjectPose bug with custom level

aouimet, 2018-07-19:
hello, I have some problem with Airsim, after activating the computer vision, all the script that I run with python works well, except the one that interests me,
objects.
 I arrive to find the position of the object, with the name or the tag, (it is in movable) by cons it can not manage to move it, simSetObjectPose does not work.

 On the other hand in the level blocks, it works very well and if I recreate a new level, I can also move the object to a new position using python. but in a custom environment, the setObject does not work.

 How could my problem be solved? Do you have any idea of ​​what might affect this command on my custom create levels?
I also have this problem with the level coming from the Market which is proposed in the tutorial of airsim (mountain landscape)

`(UnrealSourceCode) C:\Users\aouimet>python C:\Users\aouimet\Documents\Airsim_download_from_git\AirSim-master\PythonClient\computer_vision\objectsTEST.py
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

detect2 - Position: <Vector3r> {   'x_val': -135.8000030517578,
    'y_val': 39.599998474121094,
    'z_val': -2.299999952316284}, Orientation: <Quaternionr> {   'w_val': 1.0,
    'x_val': -0.0,
    'y_val': -0.0,
    'z_val': 0.0}
detect1 - Position: <Vector3r> {   'x_val': nan,
    'y_val': nan,
    'z_val': nan}, Orientation: <Quaternionr> {   'w_val': nan,
    'x_val': nan,
    'y_val': nan,
    'z_val': nan}
Non-Existent - Position: <Vector3r> {   'x_val': nan,
    'y_val': nan,
    'z_val': nan}, Orientation: <Quaternionr> {   'w_val': nan,
    'x_val': nan,
    'y_val': nan,
    'z_val': nan}
Traceback (most recent call last):
  File "C:\Users\aouimet\Documents\Airsim_download_from_git\AirSim-master\PythonClient\computer_vision\objectsTEST.py", line 45, in <module>
    success = client.simSetObjectPose("detect2", pose1, True);
  File "C:\Users\aouimet\Documents\Airsim_download_from_git\AirSim-master\PythonClient\airsim\client.py", line 108, in simSetObjectPose
    return self.client.call('simSetObjectPose', object_name, pose, teleport)
  File "C:\ProgramData\Anaconda3\envs\UnrealSourceCode\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\ProgramData\Anaconda3\envs\UnrealSourceCode\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'simSetObjectPose' with argument count 3.`

( i just try to detect the first one.''detect2'' i use Anaconda , Ue4 and AirSim for 4.18, )

jianghe01, 2018-07-20:
i wanna move objects too, do u find any way to finish the job?

aouimet, 2018-07-23:
you need to rebuild the engine from source, after that, you need to rebuild your level ,  i use anaconda - you need tornado , and mgspack ( Something like that) but i download the mgs pack manually i didnt find it on anaconda. after that it suppose to work , just open your game run airsim , open terminal into anaconda and run the script .

download airsim first , and dont forget to enable into your code the airsim plugin .

engin in 4.18, airsim in 4.18 .

sytelus, 2018-07-25:
The `simSetObjectPose` works only if target object is marked as movable in UE Editor. By default is it marked as Static. See https://docs.unrealengine.com/en-us/Engine/Actors/Mobility.

jianghe01, 2018-07-25:
> 答复: Microsoft/AirSim <reply@reply.github.com>
> 日期: 2018年7月23日 星期一 下午9:40

Thanks a lot

closed 2020-02-14 by ironclownfish