microsoft/AirSim #1266: How to show car states information in self-built environment?

BrianN92, 2018-07-19:
Hi all! Does anyone know how to show the following information in a self-built environment?  I have created a new AirSim environment in UE4, and I would like to have the same state information shown in it. Thanks for your help!
[image]

orangeccc, 2018-07-23:
I want to kown that too.

sytelus, 2018-07-24:
It should be shown just by default. If you are seeing car and able to drive it (i.e. AirSim is working) then log messages gets displayed automatically.

BrianN92, 2018-07-24:
Hi, @sytelus . Thanks for your reply. Sadly, as shown in the image, the log messages were not shown by default in my game. 
[image]

BrianN92, 2018-07-24:
In the settings.json file, I also tried to set "LogMessagesVisible" to true, which didn't help either.

sytelus, 2018-10-17:
I realized you are using release build. I think log messages are not shown in released binaries however this would be a good feature to add.