microsoft/AirSim #1269: Multiple cameras in Computer Vision mode

martijn-vinotion, 2018-07-20:
Hello everybody,

I want to be able to request images from different angles through the python API in Computer Vision mode.
In particular, I want to be able to place pawns/cameras in the editor and then request images from their angle and position.

My current approach is to define multiple vehicles that have a duplicate of the ComputerVisionPawn as their pawn which then can be placed in the editor.

Currently, this results in the following situation.
When I open the standalone game, we see the level from the viewpoint of the last vehicle that is defined while the subwindows are from cameras of the vehicle that was defined first.
When I request images, via `client.simGetImages(image_requests,'first_vehicle')`, I get the images from the viewpoint of the last added vehicle.

This raises the following questions:

1. How do I specify in which vehicle viewpoint the game starts?
2. How do I specify which vehicle viewpoints are used for the subwindows?
3. Why am I getting the wrong images from `simGetImages`?

Can anyone help me with this?

Here is my settings file:
```
{
	"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
	"SettingsVersion": 1.2,
	"SimMode": "ComputerVision",
	"ViewMode": "Fpv",
	"Recording": {
    	"RecordInterval": 1,
		"Cameras": [
			{ "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": false },
			{ "CameraName": "0", "ImageType": 5, "PixelsAsFloat": false, "Compress": false }
		]
	},
	"EngineSound": false,
	"CameraDefaults": {
		"CaptureSettings": [
			{
				"ImageType": 0,
				"Width": 640,
				"Height": 480
			}
		],
		"NoiseSettings": [
			{
				"Enabled": false,
				"ImageType": 0,

				"RandContrib": 0.2,
				"RandSpeed": 100000.0,
				"RandSize": 500.0,
				"RandDensity": 2,

				"HorzWaveContrib":0.03,			
				"HorzWaveStrength": 0.08,
				"HorzWaveVertSize": 1.0,
				"HorzWaveScreenSize": 1.0,

				"HorzNoiseLinesContrib": 1.0,
				"HorzNoiseLinesDensityY": 0.01,
				"HorzNoiseLinesDensityXY": 0.5,

				"HorzDistortionContrib": 1.0,
				"HorzDistortionStrength": 0.002
			}
		],
		"X": 0, 
		"Y": 0, 
		"Z": 0,
		"Pitch": 0, 
		"Roll": 0, 
		"Yaw": 0    
	},
	"SubWindows": [
		{"WindowID": 0, "CameraName": "default", "ImageType": 3, "Visible": true},
		{"WindowID": 1, "CameraName": "default", "ImageType": 5, "Visible": true},
		{"WindowID": 2, "CameraName": "alternative", "ImageType": 0, "Visible": true}    
	],
	"SegmentationSettings": {
		"InitMethod": "",
		"MeshNamingMethod": "",
		"OverrideExisting": false
	},
	"PawnPaths": {
		"BareboneCar": {"PawnBP": "Class'/AirSim/VehicleAdv/Vehicle/VehicleAdvPawn.VehicleAdvPawn_C'"},
		"DefaultCar": {"PawnBP": "Class'/AirSim/VehicleAdv/SUV/SuvCarPawn.SuvCarPawn_C'"},
		"DefaultQuadrotor": {"PawnBP": "Class'/AirSim/Blueprints/BP_FlyingPawn.BP_FlyingPawn_C'"},
		"DefaultComputerVision": {"PawnBP": "Class'/AirSim/Blueprints/BP_ComputerVisionPawn.BP_ComputerVisionPawn_C'"},
		"first_vehicle": {"PawnBP": "Class'/AirSim/Blueprints/CVPawn.CVPawn_C'"},
		"last_vehicle": {"PawnBP": "Class'/AirSim/Blueprints/CVPawn2.CVPawn2_C'"}
	},
	"Vehicles": {
		"first_vehicle": {
			"VehicleType": "Fpv",
			"AutoCreate": true,
			"PawnPath": "first_vehicle",
			"Cameras": {
				"default": {
					"CaptureSettings": [
						{
							"ImageType": 5,
							"Width": 640,
							"Height": 480
						}
					],
					"X": 0, 
					"Y": 0,
					"Z": 0,
					"Pitch": 0, 
					"Roll": 0, 
					"Yaw": 0    
				},
				"alternative": {
					"CaptureSettings": [
						{
							"ImageType": 5,
							"Width": 640,
							"Height": 480
						}
					],
					"X": 0, 
					"Y": 0,
					"Z": 0,
					"Pitch": 0, 
					"Roll": 0, 
					"Yaw": 180    
				}
			}
		},
		"last_vehicle": {
			"VehicleType": "Fpv",
			"AutoCreate": true,
			"PawnPath": "last_vehicle",
			"Cameras": {
				"default": {
					"CaptureSettings": [
						{
							"ImageType": 5,
							"Width": 640,
							"Height": 480
						}
					],
					"X": 0, 
					"Y": 0,
					"Z": 0,
					"Pitch": 0, 
					"Roll": 0, 
					"Yaw": 0    
				},
				"alternative": {
					"CaptureSettings": [
						{
							"ImageType": 5,
							"Width": 640,
							"Height": 480
						}
					],
					"X": 0, 
					"Y": 0,
					"Z": 0,
					"Pitch": 0, 
					"Roll": 0, 
					"Yaw": 180    
				}
			}
		}
	}
}
```

sytelus, 2018-07-24:
Currently The first vehicle you specify in "Vehicles" section becomes FPV vehicle. The sub-window shows views from FPV vehicle. You can use `"IsFpvVehicle": true` to make any vehicle as FPV however we haven't tested this so its experimental at this stage (and so undocumented).

I don't understand "default" and "alternative" sections in "Cameras". I also don't understand `"VehicleType": "Fpv"`. These are not supported.

Looks like you are spawning both vehicles at same coordinates so you might get same views from both.   

Here's the [test script](https://github.com/Microsoft/AirSim/blob/8a01d19173db0b872dda4ab453b0d476c4319508/PythonClient/multirotor/multi_agent_drone.py#L60) we use for multi-vehicle and it seems to get images correctly.

martijn-vinotion, 2018-07-24:
Dear @sytelus, 

Thanks for your reply! The "default" and "alternative" cameras aren't very crucial, I was trying to add extra cameras. Showing these in the subwindows did work but I couldn't request them through `simGetImages`, but this is not very crucial.

I was hoping that the ComputerVision 'vehicles' would spawn at the location where I placed the pawns in the editor. This did occur, as the main window was viewing from one pawn and the subwindows gave images from the perspective of the other pawns.

After I followed your steps, both 'vehicles' spawn at the location of PlayerStart (as all images are now from this perspective) so that I can't tell whether `simGetImages` gets the image from the correct vehicle.

Is it possible to have multiple vehicles spawn at different (editor-specified) locations (so without setting coordinates in the *settings.json*)?

martijn-vinotion, 2018-07-24:
After trying around for a while, I am now certain that my two vehicles are at two different locations. (`simGetVehiclePose` returns different poses) but still all images are taken from one of the vehicle, regardless of whether I specify a `vehicle_name` argument in the `simGetImages`.

Therefore, I suspect that something is going wrong inside `simGetImages` when using multiple vehicles in ComputerVision mode.

Beijingsheng, 2018-08-07:
Have you figured this out? I am also trying to add new cameras to the car. I make some changes in CarPawn.cpp and Settings.json, neither work. Can anyone please introduce the process in detail, Many thanks!!

nitin-ppnp, 2019-09-17:
> Currently The first vehicle you specify in "Vehicles" section becomes FPV vehicle. The sub-window shows views from FPV vehicle. You can use `"IsFpvVehicle": true` to make any vehicle as FPV however [...]

The provided test script uses multirotor mode, not computer vision. I want three independent moving cameras. It seems like there is only one camera in computer vision mode and it is the one which is defined in the end of the **settings** file.  Please have a look at my **settings.json** below. If I set pose of _machine_3_ then the pose of the other two are also set to be the same. Same is the case for the images from these vehicles. If I try to set the pose of the first two (i.e. _machine_1_ and _machine_2_) nothing happens (also no error).
```
{
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
    "SettingsVersion": 1.2,
    "CameraDefaults": {
                        "CaptureSettings": [
                                              {
                                                "Width": 640,
                                                "Height": 360
                                              }
                                            ]

                        },

	"Vehicles": {

	"machine_1": {
		  "VehicleType": "ComputerVision",
		  "X": 20, "Y": 0, "Z": -2
		  },
	"machine_2": {
		  "VehicleType": "ComputerVision",
		  "X": 10, "Y": 10, "Z": -2
		  },
	"machine_3": {
		  "VehicleType": "ComputerVision",
		  "X": 10, "Y": -10, "Z": -2
		  }
	},
	"SimMode" : "ComputerVision"
}
```

closed 2022-04-28 by jonyMarino