microsoft/AirSim #1272: Can't control drone after using simSetVehiclePose

Gotfeurdom, 2018-07-22:
When I use simSetVehiclePose, the drone is set at the right position (as long as there is no collision) and I can still see the rotors moving depending on my control inputs.
However, the drone is stuck at the same position.

I don't see anything missing from the description of that function here https://github.com/Microsoft/AirSim/blob/master/docs/apis.md.

What am I doing wrong ?

sytelus, 2018-07-24:
Are you using Computer Vision mode? The `simSetVehiclePose` ix meant for CV mode. There is no physics in this mode so drone won't move.

Gotfeurdom, 2018-07-24:
I'm using the default mode, which I read is not the CV mode.
How can I swap back to non-CV mode then ?

Also, how can I start from any point of my map using my python program and get the real coordinates (ie. relative to the map and not the start point) without using that setvehicle function?

alexanderkoumis, 2019-02-14:
Hi, any updates on this? I would like to reset the drone after a certain period to a position on the ground. My use case is that I would like to make the drone perform a certain sequence, collect some data, make it move back to the starting position, and repeat. If calling `simSetVehiclePose` switches the simulator to CV mode where the physics are turned off then I cannot accomplish this. Any way I can reset the vehicle pose without computer vision mode?

Thank you.

Gotfeurdom, 2019-02-14:
> Hi, any updates on this? I would like to reset the drone after a certain period to a position on the ground. My use case is that I would like to make the drone perform a certain sequence, collect [...]

I gave up using that simulator because of this
I think they were working on that functionality but I'm not sure if they implemented it yet

If you have an important project to do and cannot waste any time hacking into their simulator, I would strongly advise you to look somewhere else for an alternative

ironclownfish, 2020-02-14:
Are you enabling API control and calling takeofAsync() after setting the drone's position?

jonyMarino, 2020-10-19:
Closed due to inactivity. Doc update pending

PR #3363 (2021-02-02): Fix losing physics after setting pose

Fixes: #1272
Fixes: #1856
Fixes: #1246
Fixes: #3049

## About
This is a slightly modified version of @songxiaocheng's PR originally at #2324. This PR includes all his changes with the exception of 1 line that still has an open question.

## How Has This Been Tested?
Build was confirmed to work, and a modified version of the teleport.py was tested where a moveToPositionAsync() API call was made after a simSetVehiclePose() call.

## Screenshots (if appropriate):
closed 2020-10-19 by jonyMarino