microsoft/AirSim #1274: client.simGetSegmentationObjectID() always returns -1

martijn-vinotion, 2018-07-23:
When I request the segmentation object ID, it always returns -1. Even if I had requested
`client.simSetSegmentationObjectID('object_name',1)` (which does return `True` and does change the color of the segmentation mask) beforehand.

I am running AirSim in CV-mode.

Is anyone else also encountering this problem?

saiabinesh, 2018-08-20:
Yes, I am. Airsim latest commit. Windows 10. Custom Unreal environment. 
For instance, 
found = client.simSetSegmentationObjectID("FbxScene_GraveCart[\w]*", 22, True);
correctly changes all the carts colors to 22.  (verified by extracted image)

But 
print(client.simGetSegmentationObjectID("FbxScene_GraveCart4"))
returns -1

sytelus, 2018-10-17:
-1 is returned when said object is not found. Note that UE requires you to manually name the object (i.e. XYZ_4 is auto-generated and may not be valid in next run).

If this is still a problem, please file a new issue with code, settings and clear steps to reproduce the issue.

closed 2018-10-17 by sytelus

ykpgrr, 2019-01-15:
I've seen the same problem. Did you solve? @saiabinesh @martijn-vinotion

martijn-vinotion, 2019-01-19:
Unfortunately I did not manage to solve it. I worked around it by simply setting the segmentation ids to whatever I needed.

saiabinesh, 2019-03-12:
nope. @ykpgrr

alexander-hamme, 2019-09-03:
I can confirm that this is still an issue, I don't know why it was marked as resolved.

dnychennnn, 2020-06-23:
Could anyone verified if this is still an issue? I still only get -1 even if I set the segmentation ID to the object first. Also the seg_rgb.txt doesn't really match the segmentation output.

RachithP, 2021-02-01:
@dnychennnn, looks like #2845 solves the matching issue.

marcb19990, 2023-01-04:
Anyone found a solve? I have the same issue. I use setId and it works (visually confirmed) but the getId always returns -1.

Input:
success = self.drone.simSetSegmentationObjectID("DroneFollower1", 25, True);
print(success)

success = self.drone.simSetSegmentationObjectID("DroneFollower2", 78, True);
print(success)        

print(self.drone.simGetSegmentationObjectID("DroneFollower1"))
print(self.drone.simGetSegmentationObjectID("DroneFollower2"))

Output:

True
True
-1
-1