microsoft/AirSim #1275: Positionning a car and setting its speed and gear

Ricocotam, 2018-07-23:
Hi, I'm looking for doing imitation learning within AirSim but I encountered an issue. I'd like to isolate a turn from a circuit to do imitation learning on this. But I couldn't find a way to set the car speed (and other parameters needed) so it starts in the exact same position it was when I was driving and thus could (may be) learn.

Is it possible, if yes how ? If not how can I hack it / will it be possible and when ?

Thanks

sytelus, 2018-07-24:
Not sure what your question means. May be you want to log your steering commands and then use Python to play it back?

Ricocotam, 2018-07-24:
I'm already logging all I need. But I'd like to set the position of the car at the beginin of my turn and set its speed so I'm in the same situation. But my turn might happen 10 minutes after the start of the run and i don't want to play back 10 minutes to learn 10s of driving.

I'm not english, I sometimes have a hard time explaining myself

Gotfeurdom, 2018-07-24:
Basically he wants to set the state of his vehicle however he wants : position/orientation/speed/whatever else with a single command.

A basic functionality for a simulator ...

sytelus, 2018-10-17:
We now have `getCarState` API that returns all of the state in one shot.

If this is still a problem, please file a new issue with code, settings and clear steps to reproduce the issue.

closed 2018-10-17 by sytelus