microsoft/AirSim #1283: How to use Computer Vision mode in CityEnviron?

noNameForMe1994, 2018-07-25:
Hey,

I am trying to set simMode to Computer Vision in CityEnviron provided in airsim releases. The settings.json I used is as follows:

> {
> "SettingsVersion": 1.2,
> "SimMode":"ComputerVision",
>   "Recording": {
>     "RecordOnMove": true,
>     "RecordInterval": 0.01,
>     "Cameras": [
>         { "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": true }
>     ]
>   }
> }

However, when I started the CityEnviron, an error called 
Error at startup: 
Failed to load asset class - Class'/Airsim/Blueprint/BP_ComputerVisionPawn.BP_ComputerVisionPawn_C' 
occurs. 
[image]

Is there anyway can I use Computer Vision mode in CityEnviron?

noNameForMe1994, 2018-08-07:
Any help would be greatly appreciated!

sytelus, 2018-10-17:
This was fixed. Could you please try out latest?

closed 2018-10-17 by sytelus