microsoft/AirSim #1287: Adding Custom Vehicle Blueprint as a New SimMode inside the Plugin

appmechanic, 2018-07-26:
Hey guys , 
I Have Created a Custom Vehicle in a blueprint , Which is kind of a drone and moves on the wall of a tank in vertical direction , it has magnetic effect which keeps the Drone attached to the Wall of the tank , Now i want this drone to use as pawn from inside the AirSimPlugin , Drone has its own Controls and Camera , I want to activate its own controls and Camera when this drone is activated from the Settings.json File.
iam looking to set this drone as  a SimMode just like Car mode . 

As of now i can set the path for my Drone from settings.json file by setting pawnpaths inside this file , For this i have reparented my blueprint with CarPawn class as parent  , which leds me to possess my drone , But controls doesnt work and camera is also not from drone blueprint setups.

please help Anyone .

Beijingsheng, 2018-08-17:
I encountered the same problem. I have built my own vehicle and I want to set my own mode. Have you solved your problem? 
Could anyone please help?

shaoxiang, 2020-05-29:
@appmechanic  @Beijingsheng Hello, do you have a solution now? My requirement is add differential drive robot in Airsim. But there's no official way to do it.

closed 2021-10-19 by jonyMarino

YongzhengCui, 2023-10-13:
@shaoxiang Hi, I have the same requirements as you. May I ask if you have successfully added differential drive robots to Airsim?