microsoft/AirSim #1289: 'CameraInfo' object has no attribute 'proj_mat' when requesting simGetCameraInfo

martijn-vinotion, 2018-07-26:
I am running ComputerVisionMode with two cameras on Ubuntu 16.04.
When I call client.simGetCameraInfo(0), I get

> 'CameraInfo' object has no attribute 'proj_mat'

Has anyone else encountered this problem?
Or does anyone know how to solve this?

husha1993, 2018-07-29:
I encountered this problem too....

Mayankm96, 2018-08-05:
Any updates on this?

quachconghoang, 2018-09-01:
It happens on Windows 10 - Anaconda 5.2 -python 3.6 :(

diarmaidocualain, 2018-10-09:
Strange, it is working here for me. I am using Python 3.7, Windows 10,  Airsim Version : 1.1.7, commit fd8d5dfd62b888116ecf1819d260bf0a2a44e3cc 

```
camera_info = client.simGetCameraInfo("front_left")
print("CameraInfo %s: %s" % ("front left", pp.pprint(camera_info)))
```

```
<CameraInfo> {   'fov': 90.0,
    'pose': <Pose> {   'orientation': <Quaternionr> {   'w_val': 0.2002500295639038,
    'x_val': 0.0,
    'y_val': -0.0,
    'z_val': -0.9797448515892029},
    'position': <Vector3r> {   'x_val': 0.3910643756389618,
    'y_val': -5.46964693069458,
    'z_val': -0.5919323563575745}},
    'proj_mat': <ProjectionMatrix> {   'matrix': [   [   -4.371139183945161e-08,
                      0.0,
                      0.0,
                      0.0],
                  [   0.0,
                      -5.828185578593548e-08,
                      0.0,
                      0.0],
                  [   0.0,
                      0.0,
                      0.0,
                      1.0],
                  [   0.0,
                      0.0,
                      2.0,
                      0.0]]}}

```

BrunoXL, 2018-10-25:
Anyone found a solution for that! Its better to run my code from its own folder instead of Airsim's Python Client, for git stuff etc

zimmy87, 2021-03-12:
Tried to repro this on both windows 10 and ubuntu 18.04, and the scenario worked for me both times. From reading previous comments, it sounds like this was happening due to an outdated version in pip and is no longer an issue.

closed 2021-03-12 by zimmy87