microsoft/AirSim #1292: python moveByVelocityZ command has drone continue to move after duration of last command

c-maxey, 2018-07-26:
Hello,

When using the moveByVelocityZ (and others) command from the python api for the drone, once a commands duration has elapsed the drone still continues to move with the velocity of the last command given.  Is this normal behavior?  Or should we be excepting the drone to stop once the command duration is done?  If the former, then why have a command duration at all?

sytelus, 2018-08-10:
I believe drone should be entering in house state unless you are executing this command over and over.

lovettchris, 2018-12-18:
The problem you might be hitting is that PX4 does not automatically "hover" at the end of an "offboard" control loop.  The default is to "land" if "offboard control suddenly stops".  Setting this parameter on the PX4 may help, as it tells the PX4 to go to "loiter" mode in this case:
```
param set COM_OBL_ACT 1
```
Alternately, you could add this command after your moveByVelocityAsync to do that programmatically.  
```
self.client.hoverAsync().join()
```
Although I'd still recommend setting COM_OBL_ACT to 1 to ensure the smoothest transition between these two commands.

whxru, 2019-03-06:
@lovettchris I met the same problem but with SimpleFlight drone, and the drone still moves even if I add the hover command.

lovettchris, 2021-06-04:
I see, well this might be a bug in the SimpleFlight  implementation then....