microsoft/AirSim #1295: Unable to package on UE4.20

jeffreykxiao, 2018-07-27:
I know that 4.20 doesn't have any official support here, but I think this will cause some trouble for making future releases of AirSim. 

I've included AirSim in 4.18 and 4.19 builds without an issue (including packaging UE4 projects) but after moving to 4.20, I'm seeing a linker error that only shows up when packaging the UE4 project into a binary. The error is as follows:

```
UATHelper: Packaging (Windows (64-bit)):     Module.AirSim.cpp.obj : warning LNK4217: locally defined symbol ??1bad_cast@std@@UEAA@XZ (public: virtual __cdecl std::bad_cast::~bad_cast(void)) imported in function "public: virtual __cdecl clmdep_msgpack::v1::type_error::~type_error(void)" (??1type_error@v1@clmdep_msgpack@@UEAA@XZ)
UATHelper: Packaging (Windows (64-bit)):     Module.AirSim.cpp.obj : warning LNK4217: locally defined symbol ??0bad_cast@std@@QEAA@AEBV01@@Z (public: __cdecl std::bad_cast::bad_cast(class std::bad_cast const &)) imported in function "public: __cdecl clmdep_msgpack::v1::type_error::type_error(class clmdep_msgpack::v1::type_error const &)" (??0type_error@v1@clmdep_msgpack@@QEAA@AEBV012@@Z)
UATHelper: Packaging (Windows (64-bit)):     Module.AirSim.cpp.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl std::bad_cast::bad_cast(char const *)" (__imp_??0bad_cast@std@@QEAA@PEBD@Z) referenced in function "public: __cdecl clmdep_msgpack::v1::type_error::type_error(void)" (??0type_error@v1@clmdep_msgpack@@QEAA@XZ)
UATHelper: Packaging (Windows (64-bit)):     Module.AirSim.cpp.obj : error LNK2001: unresolved external symbol "protected: virtual void __cdecl std::bad_cast::_Doraise(void)const " (?_Doraise@bad_cast@std@@MEBAXXZ)
```

sytelus, 2018-08-03:
This is very strange but here is my initial guess. Somewhere in rpclib they are using dynamic_cast which requires linking to std library with std::bad_cast. But for some reason, linker can't find it. One reason may be mangled name. This could happen if you had compiled rpclib long time ago with different compiler and now using its binaries with another compiler version. I would suggest try fresh clone and build AirSim from there using same compiler that UE 4.20 uses.

jeffreykxiao, 2018-08-03:
Thanks for the response @sytelus. I think you're right about the reason, but I'm not sure why the symptoms continue to show up. 4.20 moved UE4 to VS2017 build tools and I tried the latest release of AirSim, followed the build instructions, and still experienced the issue.

hxdnshx, 2018-08-10:
The same error occurs when I use RpcLib outside of AirLib.
Here is my solution:
1.replace all std::bad_cast in rpclib to std::exception.
2.Recompile AirSim, and copy plugin folder to your unreal project.

jeffreykxiao, 2018-08-10:
@hxdnshx Thanks for the tip. I think that's a pretty good workaround that's working, but it sucks that we have to modify the library we're pulling in.

TritonSailor, 2018-08-23:
@hxdnshx I'm having the same problem.  Could you please provide more detail on the solution?  

Which files need to be changed? Did you modify the RpcLib files that AirSim downloads, or did you build RpcLib from source yourself?

hxdnshx, 2018-08-24:
I modified the rpclib which downloaded by AirSim(AirSim/external/rpclib/rpclib-2.2.1), then builded it. You need to copy the generated lib file to AirSim\AirLib\deps\rpclib\lib\x64\$(Configuation).

The following files might contain std::bad_cast:
 rpclib\dependencies\include\asio\generic\datagram_protocol.hpp
 rpclib\dependencies\include\asio\generic\raw_protocol.hpp
 rpclib\dependencies\include\asio\generic\seq_packet_protocol.hpp
 rpclib\dependencies\include\asio\generic\stream_protocol.hpp
 rpclib\dependencies\include\asio\ip\impl\address.ipp
 rpclib\dependencies\include\asio\ip\impl\address_v6.ipp
 rpclib\include\rpc\msgpack\v1\object_fwd.hpp

TritonSailor, 2018-08-24:
Thanks, I was able to package with 4.20.2 after editing those files.

closed 2020-01-07 by jeffreykxiao