microsoft/AirSim #1300: when i build Droneshell fatal error C1083: 无法打开包括文件: “Eigen/Dense”: No such file or directory

xiaobaichongchongchong, 2018-07-29:
when i build Droneshell fatal error C1083: 无法打开包括文件: “Eigen/Dense”: No such file or directory

xiaobaichongchongchong, 2018-07-29:
Eigen/Dense   Why can't I find this folder?

xiaobaichongchongchong, 2018-07-29:
and this  fatal error C1083: 无法打开包括文件: “rpc/rpc_error.h”: No such file or directory

AloshkaD, 2018-07-29:
I think I had a similar error in the past. I manually downloaded and copied eigen but that has solved this problem and created new problems. What solved all my problems was making sure that visual studio had VC++ 2015 installed. Check out the screen shot

[image]

xiaobaichongchongchong, 2018-08-01:
Thank you for your reply, I have solved this problem myself, I checked the code in build.cmd, according to the error prompt, I found that I can not get the zip package to download the connection. So I manually downloaded and created the folder according to the logic of the code, put the required files into it, the problem is solved, but I still don't know why the code of build.cmd can't download the zip package, if anyone sees it and Know, I hope to reply so that others can see it.

hesham-akmal, 2018-09-11:
For anyone still having this problem, the .bat files needs to be run on "x64 Native Tools Command Prompt for VS 2017/2015", use windows search to bring it up then run the build.bat

If it says cmake isn't found, download it manually from their website.

jonyMarino, 2020-08-12:
Thank you @hesham-akmal, That is in the instructions of the current documentation

closed 2020-08-12 by jonyMarino