microsoft/AirSim #1301: Camera Computer vision , Add more ?

aouimet, 2018-07-30:
hi, is it possible in the computer vision mode to add camera and receive images of these? and I would also like to know how can I move them later and / or choose which camera to talk to from python?

my second question would be to know if there is a way from python to talk to blueprint or object feature? for example, being able to activate the triggering of a rain-type blueprint (which would normally be associated with a key or a trigger or an on / off)? or any other change in this sense?

Thank you for your future answers

sytelus, 2018-08-03:
Please see settings.json for how to add more cameras to vehicle. Currently there are no APIs to trigger BPs but one can certainly implement that.

NextSim, 2018-08-03:
An API to trigger blueprints would be cool.

I'm currently simulating key presses to trigger them..

CMartinETH, 2020-03-05:
I am not sure if this issue is solved now @sytelus ? I also tried to define more cameras to get in the ComputerVision mode. So I use the same settings file which works in the Multirotor mode, but change the SimMode and the VehicleType. 
But I still cannot retrieve these camera images over the API.

Any Ideas what I am doing wrong? Or is it simply not possible to define further cameras in CV mode? 

I need the CV mode as I want to be able to move the drone around with the keyboard while I am capturing images at a certain rate in the background. If there is another possibility, that would also be fine for me.

jonyMarino, 2020-08-19:
regarding your question: I would also like to know how can I move them later and / or choose which camera to talk to from python? take a look at : https://microsoft.github.io/AirSim/image_apis/#camera-apis

closed 2020-08-19 by jonyMarino