microsoft/AirSim #1306: ERROR: clang++ and libc++

Gaduu, 2018-08-01:
Hi,

Just starting on AirSim using Ubuntu 16.04, I was following the instructions on "Build AirSim on Linux" page, got as far as setting up the airsim (setup.sh) but when I ran build.sh I get the following error: 

"The llvm-souce-50 folder was not found! Mystery indeed.
ERROR: clang++ and libc++ is necessary to compile AirSim and run it in Unreal engine
Please run setup.sh first."

Also, if it helps, whilst running the setup.sh and the very end I got:

"W: Some index files failed to download. They have been ignored, or old ones used instead."

Appreciate if anyone could help me to solve this out,

Cheers,

sytelus, 2018-08-03:
I just did the clean build and everything compiles fine. I would suggest to just recreate clone and try again. setup.sh needs to run first.

Amrtanshu, 2018-08-22:
Did you clear it out the issues?? I am having same issue right now

Gaduu, 2018-08-22:
Nothing worked, turns out it was something to do with the system so I got myself a fresh copy of elementary OS and all worked perfectly fine. Also, just FYI if you don't need to compile the source code and only use AirSim and Python API than go straight for release version of the simulation.

closed 2018-08-22 by Gaduu

deksprime, 2018-10-13:
This issue is related to broken packages where  `"W: Some index files failed to download. They have been ignored, or old ones used instead."` indicates a some kind of broken repository or something of that sort. You should find which PPA causes the problem and you should either remove it or fix it somehow. Worked for me!

jackfong123, 2019-12-03:
```
mkdir -p llvm-build/output
wget http://releases.llvm.org/4.0.1/clang+llvm-4.0.1-x86_64-linux-gnu-debian8.tar.xz
tar -xf "clang+llvm-4.0.1-x86_64-linux-gnu-debian8.tar.xz" -C llvm-build/output
```

solved the problem.