microsoft/AirSim #1310: Error running own built binary

kwea123, 2018-08-05:
System :
*  Airsim version : up-to-date (August 5th)
*  Unreal Engine : 4.18
*  Windows 10

Problem :

I want to use Airsim in my own environment, I am able to import the plugin Airsim in Unreal editor like the [tutorial](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md), and am able to drive the car inside the editor using the "Play" button. 

Next, when I package the project, it runs successfully.

However, when I launch the .exe file, it shows the same error as #1080.
I also downloaded other pre-compiled binaries from the release 1.2.0, they run without any problem.

sytelus, 2018-08-10:
Here's what is happening: Unreal packaging system tries to find what assets are being used and it is not detecting that AirSim assets are being used so its not including those assets.

Make sure you are including all maps in Package Settings. If you want to only select one map then I would suggest create a new map and drag and drop vehicle BPs from AirSim plugin on that map and include that map in packaging settings. Take a look at Packaging Settings of Blocks project to see example.

kwea123, 2018-08-11:
Thank you. I got this to work.
However, I had to copy the `AirsimAssets.umap` file under `\AirSim\Unreal\Environments\Blocks\Content` to my project and include it when packaging. Am I doing things correctly? Since this asset file can be found nowhere else... 

If that's the right way to do, I can do a doc update and pull request. Thanks!

diarmaidocualain, 2018-10-05:
Hi kwea123, I am getting the same error, "Error at startup: Failed to load asset class - Class'/Airsim/Blueprints/Bp_FlyingPawn.Bp_FlyingPawn_C' when I package the map. It runs fine with the airsim plugin in the Unreal Editor. 

In your proposed fix, how did you copy and include the AirsimAssets.umap file to your package?

closed 2019-02-19 by madratman

tiax615, 2019-03-04:
> Here's what is happening: Unreal packaging system tries to find what assets are being used and it is not detecting that AirSim assets are being used so its not including those assets.
>
> Make sure you [...]

It works very well, easy way to save this problem, Thank sytelus! @sytelus

littlefish111, 2020-11-15:
It still doesn't work