microsoft/AirSim #1314: I changed Settings.json but it doesn't really work

Beijingsheng, 2018-08-06:
I am trying to take pictures of wider FOV, so I changed FOV in settings.json, however when I ran the program, pictures taken were still of 90 degrees FOV. Here is my Settings.json. Many Thanks

{
  "SettingsVersion": 1.2,
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 256,
        "Height": 144,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "MotionBlurAmount": 0
      }
    ]
  },
  "SimMode": "ComputerVision"
}

KeyneZzz, 2018-08-06:
afaik you can set  to a larger value (maybe 150, 170) to see if your change really works. This change doesn't require a restart of Unreal. Just try it.

Beijingsheng, 2018-08-07:
@KeyneZzz  Thanks for your reply. I changed FOV to 150 and run the Airsim.sln, still no difference.

Amigoshan, 2018-08-07:
From my experience, Airsim config format differs with different Airsim versions. Could you try remove "CameraDefaults": {}? I mean move "CaptureSettings": [] out of "CameraDefaults".

jinlukang1, 2018-08-08:
I want to know if you can modify the Width and the height of the recorded data, is the AirSim usable?

because when I modify my setting.json, I still get the 256*144 data.

I don't know why my setting.json doesn't work.

`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0,
  "SimMode": "Car",
    "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.5,
    "Cameras": [
        { "CameraName": "0", "ImageType": 6, "PixelsAsFloat": false, "Compress": true }
    ]
	},
    "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 512,
        "Height": 288
      }
    ]
	}
}`

Can somebody help me figure it out?

Thanks a lot.

kwea123, 2018-08-08:
@jinlukang1 Try specifying the "ImageType" in "CaptureSettings", like this:
```
"CaptureSettings": [
      { "ImageType": 6,
        "Width": 512, 
        "Height": 288
      }
]
```
It works for me.

jinlukang1, 2018-08-09:
@kwea123 Thanks s lot!
You have solved my problem!

And have you used ComputerVision mode? When I change the mode to ComputerVision mode in setting.json, the program will always exit with an error.

Beijingsheng, 2018-08-09:
Thanks, everyone, my problem has been solved. Besides, I found that when FOV is over 180, the picture taken get weird. Lower 180 works fine.

closed 2018-08-09 by Beijingsheng