microsoft/AirSim #1316: How to communicate between multi-drone in the simulation?

804476261, 2018-08-06:
I want to realize a function to communicate with other drones in multi-drone simulation, how should I do?
I use the code just like:
client1.enableApiControl(True, "Drone1")
client2.enableApiControl(True, "Drone2")
How to use client1 to send a message to client2 ??

madratman, 2018-12-21:
closing as the script answers the question pretty much. please re-open if you have more questions

closed 2018-12-21 by madratman

AIKUUM, 2020-12-26:
> I want to realize a function to communicate with other drones in multi-drone simulation, how should I do?
> I use the code just like:
> client1.enableApiControl(True, "Drone1")
> client2.enableApiControl(True, "Drone2")
> How to use client1 to send a message to client2 ??

Hi @804476261, were you able to do this? I want to do something similar

AIKUUM, 2020-12-30:
> closing as the script answers the question pretty much. please re-open if you have more questions

Hi @madratman, The script  https://github.com/Microsoft/AirSim/blob/master/PythonClient/multirotor/multi_agent_drone.py doesnt answer anything about communication between 2 drones

xxEoD2242, 2020-12-30:
@AIKUUM What you're talking about, natively in AirSim, would require you writing servers and clients (ie how this works in the real world) into AirSim or building this through Python. If you want to do that, I'd recommend messaging the maintainers for helpful hints on where to start editing the source code.

Just think of the 2 drones as 2 different computers and then implement message passing however you want (TCP, UDP, etc.). I know of nothing currently in AirSim that does what you want.

AIKUUM, 2020-12-31:
> @AIKUUM What you're talking about, natively in AirSim, would require you writing servers and clients (ie how this works in the real world) into AirSim or building this through Python. If you want to [...]

Hi @xxEoD2242, Can you suggest me some maintainers ?

Jenny-ai-uav, 2021-10-05:
@AIKUUM Did YOU find out how to communicate the 2 drones or more?
I am stuck in this problem now.
I can successfully control them but I don't know how to communicate them.
I saw mavlink, and SITL, should I need any hardware to do this? Like PX4 or something....

yo3phra, 2026-02-26:
> [@AIKUUM](https://github.com/AIKUUM) Did YOU find out how to communicate the 2 drones or more? I am stuck in this problem now. I can successfully control them but I don't know how to communicate them. I saw mavlink, and SITL, should I need any hardware to do this? Like PX4 or something....

I currently have the same problem, did you find ways to resolve it ?