microsoft/AirSim #1321: Error when copy the environment from windows back to Linux

pienrojp, 2018-08-09:
The project could not be compiled. Would you like to open it in Null Source Code Access?
Hi, I built and compiled landscapemountain in Windows and copied back to Linux.
However, when I open the file in Linux I get this error.

```
> > Running /home/pienrojp/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe LandscapeMountains Development Linux -project="/media/pienrojp/0B8104FD0B8104FD/LandscapeMountains/LandscapeMountains.uproject" -editorrecompile -progress -NoHotReloadFromIDE
> > Scale at X=65.000000, Y=24.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
> > Scale at X=640.000000, Y=512.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
> > Scale at X=640.000000, Y=512.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
> > Setting swap interval to 'Immediate'
> > 
> > Running Mono...
> > 
> > ~/UnrealEngine/Engine ~/UnrealEngine/Engine/Binaries/Linux
> > Using clang version '3.9.1' (string), 3 (major), 9 (minor), 1 (patch)
> > Using bundled libc++ standard C++ library.
> > Using fast way to relink  circularly dependent libraries (no FixDeps).
> 
```

Anyone know how to fix this?

Thanks in advance!

husha1993, 2018-08-12:
Can you upload the log file  and check the error?

pienrojp, 2018-08-13:
Thank you for your reply! This is the error that showed in terminal while compiling.
```

ERROR: UBT ERROR: Failed to produce item: /media/pienrojp/0B8104FD0B8104FD/LandscapeMountains/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so
Total build time: 1819.66 seconds (Local executor: 0.00 seconds)
[2018.08.13-12.44.10:605][651]LogLinux: Scale at X=65.000000, Y=24.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
[2018.08.13-12.44.10:607][651]LogLinux: Scale at X=635.000000, Y=495.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
[2018.08.13-12.44.10:609][651]LogLinux: Scale at X=640.000000, Y=512.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
[2018.08.13-12.44.10:805][651]LogLinux: Setting swap interval to 'Immediate'

```

husha1993, 2018-08-14:
Can you upload the log file under the UEprojectdir/Saved/Logs?
It may give more detailed error description.

pienrojp, 2018-08-18:
I was able to compile it yesterday by replacing the Plugins folder again with the one built on Linux. Then it just worked! Thank you very much!

closed 2018-08-18 by pienrojp