microsoft/AirSim #1324: Change default of each camera

Beijingsheng, 2018-08-09:
"Cameras": {   
        //same elements as CameraDefaults above, key as name
      }--
What does "key as name" means? Could someone give an example?

sytelus, 2018-08-09:
It just means child elements are same as child elements of "CameraDefaults" element.

Example:

```
"Vehicles": {
	"SimpleFlight": {
	  "VehicleType": "SimpleFlight",
	  "Cameras": {   
			"CaptureSettings": [
			  {
				"ImageType": 0,
				"Width": 256,
				"Height": 144,
				"FOV_Degrees": 90,
				"AutoExposureSpeed": 100,
				"AutoExposureBias": 0,
				"AutoExposureMaxBrightness": 0.64,
				"AutoExposureMinBrightness": 0.03,
				"MotionBlurAmount": 0,
				"TargetGamma": 1.0,
				"ProjectionMode": "",
				"OrthoWidth": 5.12
			  }
			],
			"NoiseSettings": [
			  {
				"Enabled": false,
				"ImageType": 0,

				"RandContrib": 0.2,
				"RandSpeed": 100000.0,
				"RandSize": 500.0,
				"RandDensity": 2,

				"HorzWaveContrib":0.03,			
				"HorzWaveStrength": 0.08,
				"HorzWaveVertSize": 1.0,
				"HorzWaveScreenSize": 1.0,
				
				"HorzNoiseLinesContrib": 1.0,
				"HorzNoiseLinesDensityY": 0.01,
				"HorzNoiseLinesDensityXY": 0.5,
				
				"HorzDistortionContrib": 1.0,
				"HorzDistortionStrength": 0.002
			  }
			],
			"Gimbal": {
			  "Stabilization": 0,
			  "Pitch": NaN, "Roll": NaN, "Yaw": NaN
			},
			"X": NaN, "Y": NaN, "Z": NaN,
			"Pitch": NaN, "Roll": NaN, "Yaw": NaN 
		}
	}
}
```

Beijingsheng, 2018-08-10:
I see! Many thanks!

closed 2018-08-10 by Beijingsheng

Beijingsheng, 2018-08-10:
@sytelus  I still have one question: Does "MyCamera1" means a new created camera or does it represent the default camera, which is front_center of car?

Beijingsheng, 2018-08-10:
This is how my settings.json is like:
{
  "SettingsVersion": 1.0,
  "SimMode": "Car",
  "CameraDefaults": {

    "front_left": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 256,
          "Height": 144,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "AutoExposureBias": 0,
          "AutoExposureMaxBrightness": 0.64,
          "AutoExposureMinBrightness": 0.03,
          "MotionBlurAmount": 0,
          "TargetGamma": 1.0,
          "ProjectionMode": "",
          "OrthoWidth": 5.12
        }
      ],
      "NoiseSettings": [
        {
          "Enabled": false,
          "ImageType": 0,

          "RandContrib": 0.2,
          "RandSpeed": 100000.0,
          "RandSize": 500.0,
          "RandDensity": 2,

          "HorzWaveContrib": 0.03,
          "HorzWaveStrength": 0.08,
          "HorzWaveVertSize": 1.0,
          "HorzWaveScreenSize": 1.0,

          "HorzNoiseLinesContrib": 1.0,
          "HorzNoiseLinesDensityY": 0.01,
          "HorzNoiseLinesDensityXY": 0.5,

          "HorzDistortionContrib": 1.0,
          "HorzDistortionStrength": 0.002
        }
      ],
      "Gimbal": {
        "Stabilization": 1,
        "Yaw": -90
      }
    },
    "front_right": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 256,
          "Height": 144,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "AutoExposureBias": 0,
          "AutoExposureMaxBrightness": 0.64,
          "AutoExposureMinBrightness": 0.03,
          "MotionBlurAmount": 0,
          "TargetGamma": 1.0,
          "ProjectionMode": "",
          "OrthoWidth": 5.12
        }
      ],
      "NoiseSettings": [
        {
          "Enabled": false,
          "ImageType": 0,

          "RandContrib": 0.2,
          "RandSpeed": 100000.0,
          "RandSize": 500.0,
          "RandDensity": 2,

          "HorzWaveContrib": 0.03,
          "HorzWaveStrength": 0.08,
          "HorzWaveVertSize": 1.0,
          "HorzWaveScreenSize": 1.0,

          "HorzNoiseLinesContrib": 1.0,
          "HorzNoiseLinesDensityY": 0.01,
          "HorzNoiseLinesDensityXY": 0.5,

          "HorzDistortionContrib": 1.0,
          "HorzDistortionStrength": 0.002
        }
      ],
      "Gimbal": {
        "Stabilization": 1,
        "Yaw": 90
      }
    }
  }
}

I change Yaw to -90 and -90 of front_left and front_right camera, I suppose they are going to take picture of opposite direction however neither of them change the direction. Could anyone please tell me what I write wrong in this settings.json??

muduimushui, 2019-02-27:
> This is how my settings.json is like:
> {
> "SettingsVersion": 1.0,
> "SimMode": "Car",
> "CameraDefaults": {
>
> ```
> "front_left": {
>   "CaptureSettings": [
>     {
>       "ImageType": 0,
>       "Width": [...]

hello,I have met the same problem as yours.Did you solve it ?Looking forward to your reply.