microsoft/AirSim #1325: Is it able to add a static camera on the ground?

KeyneZzz, 2018-08-10:
I want a camera that can be placed on the ground and show its captured image in subwindow when my drone is in the air. Is there any way?

sytelus, 2018-08-10:
Yes, take a look at this: https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#viewmode

KeyneZzz, 2018-08-14:
modifying viewmode seems only changes the main screen camera, but I wanna change it in subwindow.

I've find the way of modifying FlyingPawn.cpp to make it

1.add `#include "Engine/World.h"`  on the top

2.in `void AFlyingPawn::setupComponentReferences` add following codes
```
FActorSpawnParameters ap_camera_spawn_params;
static_camera_spawn_params.SpawnCollisionHandlingOverride = ESpawnActorCollisionHandlingMethod::AdjustIfPossibleButAlwaysSpawn;
gnd_camera_1_ = this->GetWorld()->SpawnActor<APIPCamera>(pip_camera_class_, (100,100,100), (90,90,0), static_camera_spawn_params);
```
(100,100,100) is the unreal location while (90,90,0) is the camera rotations

3.in `void AFlyingPawn::initializeForBeginPlay`  add the new camera
```
    std::vector<APIPCamera*> cameras = {fpv_camera_front_center_, fpv_camera_front_right_, fpv_camera_front_left_, 
        fpv_camera_bottom_center_, fpv_camera_back_center_,gnd_camera_1_};
```
build and run

then when you choose subwindow camera in settings, the 6th camera is the static camera you added

KeyneZzz, 2018-08-14:
the solution above is under ver1.1.10, cause i'm still using. and i'm trying to adapt it to ver1.2, but may take some time

KeyneZzz, 2018-08-14:
the way above is under ver1.1.10. in ver 1.2 seems changing ImageType to -1 works (not tried)

jonyMarino, 2020-10-19:
closed due to inactivity. Doc update pending

closed 2020-10-19 by jonyMarino