microsoft/AirSim #1326: ROS package for AirSim using PythonAPIs

Mayankm96, 2018-08-10:
It has been a while since the python APIs have been out along with the ROS examples of using them in nodes. I have been trying to write a complete package in ROS which links all the APIs to standard ROS messages and can be used directly into your `catkin workspace`. 

It is still under development since I have not been able to get the time to read all the APIs that are there, but the code is already open sourced: [`airsim_ros_wrapper`](https://github.com/Mayankm96/airsim_ros_wrapper). Right now I have written it for `MultirotorClient` only, but plan to extend it for `CarClient` as well.

Contributions and suggestions are really welcomed for it. Once the package is good enough, I would love to send in PR for the same. :)

sytelus, 2018-08-10:
This excellent! We look forward for your PR. I would also suggest to take a look at below which we have already merged:

https://github.com/Microsoft/AirSim/pull/1135

Mayankm96, 2018-08-16:
@sytelus I was trying to make a node for trajectory following using [MultiDOFJointTrajectory](http://docs.ros.org/melodic/api/trajectory_msgs/html/msg/MultiDOFJointTrajectory.html) messages. However, I noticed that in the python APIs while there is a position controller, there isn't any API which simultaneously sets the pose and the orientation of the drone.

Any idea on how to work around this?

madratman, 2019-05-13:
Please see [airsim_ros_pkgs](https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_ros_pkgs) and [airsim_tutorial_pkgs](https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_tutorial_pkgs) for a ros wrapper over the multirotor API
Closing coz of lack of activity. @Mayankm96 thanks for your effort on this earlier, and please feel free to contribute over the current version.

closed 2019-05-13 by madratman

machenxiang, 2019-07-26:
> Please see [airsim_ros_pkgs](https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_ros_pkgs) and [...]

when do you updata ros again?