microsoft/AirSim #1327: Change the direction of default camera

Beijingsheng, 2018-08-12:
I'm trying to change the front_left camera to left instead of facing the front, so I follow the instruction in settings.md and this is how I changed my settings.json:
"CameraDefaults": {
    "front_left": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 256,
          "Height": 144,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 100,
          "AutoExposureBias": 0,
          "AutoExposureMaxBrightness": 0.64,
          "AutoExposureMinBrightness": 0.03,
          "MotionBlurAmount": 0,
          "TargetGamma": 1.0,
          "ProjectionMode": "",
          "OrthoWidth": 5.12
        }
      ],
      "NoiseSettings": [
        {
          "Enabled": false,
          "ImageType": 0,

          "RandContrib": 0.2,
          "RandSpeed": 100000.0,
          "RandSize": 500.0,
          "RandDensity": 2,

          "HorzWaveContrib": 0.03,
          "HorzWaveStrength": 0.08,
          "HorzWaveVertSize": 1.0,
          "HorzWaveScreenSize": 1.0,

          "HorzNoiseLinesContrib": 1.0,
          "HorzNoiseLinesDensityY": 0.01,
          "HorzNoiseLinesDensityXY": 0.5,

          "HorzDistortionContrib": 1.0,
          "HorzDistortionStrength": 0.002
        }
      ],
      "Gimbal": {
        "Stabilization": 1,
        "Yaw": -90
      }
    }
However this is not working, could anyone point out the mistake of the code??

Beijingsheng, 2018-08-12:
UE4 crashed when my settings.json is like this: 

{
  "RecordUIVisible": true,
  "LogMessagesVisible": true,
  "RpcEnabled": true,
  "EngineSound": true,
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
      {
        "CameraName": "0",
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      }
    ]
  },
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 256,
        "Height": 144,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      }
    ],
    "NoiseSettings": [
      {
        "Enabled": false,
        "ImageType": 0,

        "RandContrib": 0.2,
        "RandSpeed": 100000.0,
        "RandSize": 500.0,
        "RandDensity": 2,

        "HorzWaveContrib": 0.03,
        "HorzWaveStrength": 0.08,
        "HorzWaveVertSize": 1.0,
        "HorzWaveScreenSize": 1.0,

        "HorzNoiseLinesContrib": 1.0,
        "HorzNoiseLinesDensityY": 0.01,
        "HorzNoiseLinesDensityXY": 0.5,

        "HorzDistortionContrib": 1.0,
        "HorzDistortionStrength": 0.002
      }
    ],
    "Gimbal": {
      "Stabilization": 0,
      "Pitch": 0,
      "Roll": 0,
      "Yaw": 0
    },
    "X": 0,
    "Y": 0,
    "Z": 0,
    "Pitch": 0,
    "Roll": 0,
    "Yaw": 0
  },
  "Vehicles": {
    "PhysXCar": {
      "VehicleType": "PhysXCar",
      "DefaultVehicleState": "",
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "RC": {
        "RemoteControlID": -1
      },
      "Cameras": {
        "back": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 256,
              "Height": 144,
              "FOV_Degrees": 120,
              "AutoExposureSpeed": 100,
              "AutoExposureBias": 0,
              "AutoExposureMaxBrightness": 0.64,
              "AutoExposureMinBrightness": 0.03,
              "MotionBlurAmount": 0,
              "TargetGamma": 1.0,
              "ProjectionMode": "",
              "OrthoWidth": 5.12
            }
          ]
        }
      }
    }
  }
}

madratman, 2019-02-19:
Hey, a better option is to change the placement of the camera as explained here https://github.com/Microsoft/AirSim/issues/1770#issuecomment-462949223

closed 2019-02-19 by madratman