microsoft/AirSim #1329: UE crashed when I had changed settings.json and open run.bat

Beijingsheng, 2018-08-13:
This is the example from settings.md, however, it causes a crash when I run UE4 CityEnviron. Maybe someone can explain this?
{
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "Vehicles": {
    "PhysXCar": {
      "VehicleType": "PhysXCar",
      "Cameras": {
        "front-center": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "FOV_Degrees": 120
            }
          ]
        }
      }
    }
  }
}

Beijingsheng, 2018-08-14:
There seems to be a bug of adding keys under "cameras" in "vehicle", every time I add "mycamera" in "cameras" or use defaults cameras like "front-center", UE crashes.

sytelus, 2018-10-17:
We can't reproduce this in our latest version. If problem still persist, please file new issue with code, settings and clear steps to reproduce it.

closed 2018-10-17 by sytelus