microsoft/AirSim #1330: Depth calculation based on DepthVis pfm

LunaXia, 2018-08-13:
Hello, I have recorded DepthVis pfm by modifying the settings.json file. I set "PixelsAsFloat" and "Compressed" to be true. One screenshot of the pfm opened with PfmPad is shown below. Three numbers were shown in the bottom(pointed by the red arrow) when I hovered the cursor over the image. The floating numbers shown in the bottom seem to be related to the depth information but I am not sure what the relation exactly is, i.e, how do I calculate the true depth from the floating number?  Can anyone help me with this?

In addition, I would like to modify the source code that defines how the depth information is obtained. However, I could not find the file that contains the function to get the depth information. Does anyone know which file I should look into? Thanks!
[image]

ROCSAFE, 2018-08-13:
>  The floating numbers shown in the bottom seem to be related to the depth information but I am not sure what the relation exactly is, i.e, how do I calculate the true depth from the floating number? Can anyone help me with this?

-I'm not sure exactly how to do this/how might work, but I've looked into doing something similar. Issue #154 gives some insight, but I think some of the code has been refactored since then. 

The object segmented images and depth images are calculated using stencils in material blueprints. 
[image]

I think the actual calculation is done in the SceneDepthVisMaterial blueprint:
[image]

This [answer](https://answers.unrealengine.com/questions/723922/scene-depth-capture-does-not-work.html) may help you understand where the figures come from also. It seems that SceneDepth returns the distance (in cm) from the camera, so the frac part of the sceneDepthVis material seems to be doing something like bucketing the figures and then assigning a color according to which bucket the pixel values are in.

DavidLSmyth, 2018-08-13:
>  The floating numbers shown in the bottom seem to be related to the depth information but I am not sure what the relation exactly is, i.e, how do I calculate the true depth from the floating number? Can anyone help me with this?

-I'm not sure exactly how to do this/how might work, but I've looked into doing something similar. Issue #154 gives some insight, but I think some of the code has been refactored since then. 

The object segmented images and depth images are calculated using stencils in material blueprints. 
[image]

I think the actual calculation is done in the SceneDepthVisMaterial blueprint:
[image]

This [answer](https://answers.unrealengine.com/questions/723922/scene-depth-capture-does-not-work.html) may help you understand where the figures come from also. It seems that SceneDepth returns the distance (in cm) from the camera, so the frac part of the sceneDepthVis material seems to be doing something like bucketing the figures and then assigning a color according to which bucket the pixel values are in.

husha1993, 2018-11-14:
> The floating point values are depth in meters.
Those floating points are numbers within range 0-1,
it seems that the return from depthperspective has been post-processed.
How can I obtain the real depth from processed depth values?

jonyMarino, 2020-10-19:
Closed due to inactivity. Doc update pending

closed 2020-10-19 by jonyMarino