microsoft/AirSim #1332: simGetObjectPose does not return all objects and always something different

phildue, 2018-08-14:
I try to retrieve object poses via the API. My objects are called "object0", "object1" and so on.  I give this name in the editor as well as as a tag name. I start the simulation in Unreal Editor once and run my application multiple times. So it connects retrievese the object poses and prints them.

The output is different everytime. Most of the times nothing is returned, sometimes all objects are found and sometimes different subsets of the objects are found.

sytelus, 2018-08-15:
Are these dynamic objects? Did you renamed these objects in editor? The code isn't doing any magic - it just calls Unreal API to search for objects. It might be good idea to put a breakpoint and see if UE APIs are returning the objects.

phildue, 2018-08-16:
Thanks for reply. So I was just using a cylinder object and I renamed them in the editor. Now I created a static mesh and converted it into a blueprint and with those objects it is working again. So I'm not sure if this is a problem on airsim side or whether there was a weird internal naming of the editor.

RVBldr, 2018-11-21:
Internal VSTF 7420

madratman, 2019-02-19:
closing at this was resolved, but we need to update the doc

note to self: this is a common issue questions and should be added in the docs PR

closed 2019-02-19 by madratman