microsoft/AirSim #1333: Fixes for Compiling in Linux

katetolstaya, 2018-08-15:
Below are some temporary fixes to compile Blocks in Linux. In later versions of AirSim, these should hopefully be unnecessary. 

**Error:**
When compiling AirSim, can't include rpclib and 
"#include "Eigen/Dense.h" 

**Fix**:
Either change this import to 
"#include <eigen3/Eigen/Dense.h>
or create a symlink from Eigen/ to eigen3/Eigen
And import the system headers for rpclib by changing the quotation marks "" to brackets <>

**Error:**
When updating between versions of AirSim:
file '/usr/include/linux/version.h' has been modified since the precompiled header

**Fix:** 
Delete the Unreal header file in the error and recompile Unreal. Should be quick. Alternatively you can clean and rebuild all of Unreal, but that's unnecessary. 

**Error:** 
When opening the environment in the Editor:

  error: exception handling was disabled in PCH file but is currently enabled
  error: definition of macro 'PLATFORM_EXCEPTIONS_DISABLED' differs between the precompiled header ('1') and the command line ('0')
  2 errors generated.

**Fix:**
diff --git a/Unreal/Environments/Blocks/Source/Blocks/Blocks.Build.cs b/Unreal/Environments/Blocks/Source/Blocks/Blocks.Build.cs

--- a/Unreal/Environments/Blocks/Source/Blocks/Blocks.Build.cs​
+++ b/Unreal/Environments/Blocks/Source/Blocks/Blocks.Build.cs​

@@ -6,8 +6,6 @@ public class Blocks : ModuleRules​
 {​
     public Blocks(ReadOnlyTargetRules Target) : base(Target)​
     {​
**-        PCHUsage = PCHUsageMode.UseExplicitOrSharedPCHs;​
-        bEnableExceptions = true;​**
         PublicDependencyModuleNames.AddRange(new string[] { "Core", "CoreUObject", "Engine", "InputCore" });​

     }​

 }​

diff --git a/Unreal/Plugins/AirSim/Source/AirBlueprintLib.h b/Unreal/Plugins/AirSim/Source/AirBlueprintLib.h​

**--- a/Unreal/Plugins/AirSim/Source/AirBlueprintLib.h​
+++ b/Unreal/Plugins/AirSim/Source/AirBlueprintLib.h​**

@@ -16,6 +16,7 @@​
 #include "Kismet/KismetStringLibrary.h"​​
 #include "Runtime/Landscape/Classes/LandscapeComponent.h"​
**+#include "Runtime/Engine/Classes/Engine/World.h"​**
 #include "common/AirSimSettings.hpp"​
 #include <string>​
 #include <regex>​

katetolstaya, 2018-08-15:
Outline of steps to open and package environments in Linux:
To open Unreal Editor
1. cd UnrealEngine 
2. ./Engine/Binaries/Linux/UE4Editor
3. Select desired project and open
To package binary:
1. In editor, File > Package Project
One packaging is done:
1. cd LinuxNoEditor
2. .\Blocks.sh (or alternate environment name)

katetolstaya, 2018-08-15:
To package without using the UE4 Editor GUI, you can use a bash script with something like this:

TOOLPATH="/home/USER/UnrealEngine/Engine/Build/BatchFiles/RunUAT.sh"
PROJECTPATH="../Documents/Unreal Projects/AirSimEnvNH/AirSimNH.uproject"
OUTPATH="../Documents/Unreal Projects/Binaries/"

$TOOLPATH BuildCookRun -project="${PROJECTPATH}" -noP4 -platform=Linux -clientconfig=Development -cook -build -stage -pak -archive -archivedirectory="${OUTPATH}"  -utf8output -compressed -prereqs

madratman, 2019-02-19:
We'll add these to the docs. Thanks Kate!

closed 2019-02-19 by madratman