microsoft/AirSim #1337: PX4 SITL Firmware Start Error - No Such File or Directory

ganscool86, 2018-08-17:
Hi,
I am trying to setup PX4 SITL setup for Airsim. I followed the procedure stated here. First I installed the prerequisites and then pulled the PX4 firmware code and build it using the following commands:
git clone https://github.com/PX4/Firmware.git
cd Firmware
git submodule update --init --recursive
make posix_sitl_default

The build went well without any error message. But when I try to start the PX4 firmware using the following command:
./build/posix_sitl_default/px4 ./posix-configs/SITL/init/ekf2/iris
 or
 ./build_posix_sitl_default/src/firmware/posix/px4 ./posix-configs/SITL/init/lpe/iris

It just shows No such file or directory. Can someone tell me what i am missing and how to get it to work. Thanks.

accasio, 2018-09-17:
I'm also seeing this, not familiar with PX4 so it would be great if someone could advise on what the right command is (px4 is located in build/posix_sitl_default/bin in the latest version). I created and copied the iris file in the ekf2 folder from a previous PX4 commit. 

Running ./build/posix_sitl_default/bin/px4 ./posix-configs/SITL/init/ekf2/iris gives the following output: 

INFO  [Unknown] Creating symlink ./posix-configs/SITL/init/ekf2/iris -> /home/james/Downloads/PX4/Firmware/etc
ERROR [Unknown] Error opening startup file, does not exist: etc/init.d/rcS

markusortlieb, 2018-10-15:
You can copy the folder /ROMFS/px4fmu_common to the /etc directory. That should solve the error "ERROR [Unknown] Error opening startup file, does not exist: etc/init.d/rcS"

gonzaloaguilarjimenez, 2018-11-02:
Hello, 

It worked for me as per #10254 comment from hamishwillee

make posix_sitl_default none_iris 
# Replaces: ./build/posix_sitl_default/px4 ./posix-configs/SITL/init/ekf2/iris

Hope it helps.

rohitmohan7, 2018-11-28:
Hello, 
./build/posix_sitl_default/bin/px4 ./posix-configs/SITL/init/ekf2/iris gives the following output with rCS file in etc/init.d (had to delete the existing etc file in the folder to make a new etc folder) :

157 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d/rcS 0
mount: only root can use "--types" option
etc/init.d/rcS: 107: etc/init.d/rcS: sercon: not found
etc/init.d/rcS: 112: etc/init.d/rcS: ver: not found
etc/init.d/rcS: 119: etc/init.d/rcS: uorb: not found
etc/init.d/rcS: 124: etc/init.d/rcS: tone_alarm: not found
mount: only root can use "--types" option
etc/init.d/rcS: 146: etc/init.d/rcS: tune_control: not found
etc/init.d/rcS: 147: etc/init.d/rcS: mkfatfs: not found
etc/init.d/rcS: 163: etc/init.d/rcS: ver: not found
etc/init.d/rcS: 166: [: =: unexpected operator

tried make posix_sitl_default none_iris and it dosnt detect the RCS File...…..Help Please !

madratman, 2018-12-19:
This was fixed by @clovett recently. 
Can you guys check the updated docs [px4_build](https://github.com/Microsoft/AirSim/blob/master/docs/px4_build.md) and [px4_sitl](https://github.com/Microsoft/AirSim/blob/master/docs/px4_sitl.md)

it's a `make posix_sitl_ekf2`, followed by `make posix_sitl_ekf2  none_iris`.

lovettchris, 2018-12-19:
And once you've done this make command line once, you can do this subsequently for a slightly quicker launch:
```
./build/posix_sitl_ekf2/px4 . ./posix-configs/SITL/init/ekf2/iris
```

gonzaloaguilarjimenez, 2018-12-19:
I've just tried it and worked for me.

Thanks a lot.

closed 2018-12-19 by madratman