microsoft/AirSim #1338: Remove unnecessary null pointer checks

elfring, 2018-08-17:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
* [DIJT_SAFE_DELETE](https://github.com/Microsoft/AirSim/blob/65f5528a1b2fdab8218fb3902c39851b5ceddbce/Unreal/Plugins/AirSim/Source/SimJoyStick/DirectInputJoyStick.cpp#L44 "DIJT_SAFE_DELETE macro")
* [StreamBuffer](https://github.com/Microsoft/AirSim/blob/98dd2836a73e94bb7f540edcfb12f005c9bdd9a1/MavLinkCom/MavLinkTest/Commands.cpp#L1369 "Destructor for StreamBuffer")

jonyMarino, 2020-08-20:
I think is ok to allow this kind of code. It can be a developer preference on the clarity of his code

closed 2020-08-20 by jonyMarino

elfring, 2020-08-20:
How do you think about to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability-delete-null-pointer.html "Deletion of unnecessary null pointer checks")?