microsoft/AirSim #1339: Async API for the multi-rotor demo

katetolstaya, 2018-08-17:
If I run the multirotor demo with the Blocks environment on Linux, the first time it works fine. After I close everything and re-run the demo for a second or third time, I observe that some of the async calls do not go through. For example, one quadrotor will stay on the ground without taking off. The problems get worse and worse if you try to re-run the demo.  Adding pauses after each async library call helps somewhat. 
This is with the Airsim version as of 8/15.

sytelus, 2018-08-23:
Are you closing the UE editor and restarting it? There is nothing in code that saves the state in between so this would be very strange. Also I haven't seen personally happen this. If possible, it might be good if you can state the exact steps or post video.

katetolstaya, 2018-08-27:
The exact steps are:
1) ./Blocks.sh
2) python2.7 multi_agent_drone.py
Then, I press a key to have the drones take off, and one stays on the ground, and the other takes off. 

I can't reproduce this behavior with Python 3.6 though. I believe everything works. 
Are there any known differences in LibRPC for Python 2.7 versus Python 3.6?

madratman, 2018-12-18:
This was resolved over internal discussions

closed 2018-12-18 by madratman