microsoft/AirSim #1340: yaw problem

KeyneZzz, 2018-08-21:
in airsim my drone is heading exactly at north, but the hitl output GPS says that the yaw is around 10 degrees. i think it may related to ekf2 estimator. however, since it's in hitl, is there any way to ignore the deviation?
thx

madratman, 2018-12-18:
@clovett do you think this is fixed with the recent commits?

clovett, 2018-12-19:
Unfortunately no.  I call this "crabbing" the drone has a small yaw delta that stops it from point forward in whatever path it is flying when you use YawMode(False,0).  It should be pointing directly forwards along the path it is flying, but it is not.  It has done this for a long time, and still does this - but the behavior is random, sometimes it doesn't do this at all and I have not figured out the cause.   I suspect it might have something to do with differences in the calculation of magnetic declination between what AirSim is sending to the PX4 and what the ekf2 is calculating based on given GPS locations.

sytelus, 2018-12-19:
Probably relevant issue: 
https://github.com/PX4/ecl/issues/30

So looks like PX4 has setting for declination and it’s default is probably valid for equator. If you change this setting in PX4 then you might be able to set correct latitude.

madratman, 2018-12-19:
[this](https://docs.px4.io/en/advanced_config/parameter_reference.html) is the page to look at (I am just doing a ctrl +f "declination")

EKF2_MAG_DECL is defaulted to 0 deg
EKF2_DECL_TYPE is a bitmask setting variable which allows us to lookup the declination from gps from the geo_lookup library if it's set to 0. By default, it's set to 7 though
Chris
So, should we add this to our px4 docs? @clovett, can you confirm what EKF2_DECL_TYPE is on your setup?

lovettchris, 2018-12-19:
Turns out this is an AirSim bug, the declination doesn't match what PX4 is computing so when PX4 tries to remove the declination that AirSim added, it is over compensating.  Until this is fixed it is recommended you set a geo position on the equator.  If you must fly somewhere else, then find out what declination AirSim is generating then edit the PX4 startup config file “build/posix_sitl_ekf2/tmp/posix-configs/SITL/init/ekf2/iris” and set EKF2_MAG_TYPE to 0 and EKF2_MAG_DECL to whatever AirSim is providing, and then they will cancel out nicely and the drone will fly nice and straight.