microsoft/AirSim #1342: Vehicle API for 'Car1' is not available.

EricJin101, 2018-08-22:
when I run
`python multi_agent_car.py`
I got this result

> Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "multi_agent_car.py", line 31, in <module>
    client.enableApiControl(True, "Car1")
  File "/home/eric/AirSim/PythonClient/airsim/client.py", line 37, in enableApiControl
    return self.client.call('enableApiControl', is_enabled, vehicle_name)
  File "/home/eric/anaconda3/lib/python3.6/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/eric/anaconda3/lib/python3.6/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'Car1' is not available. This could either because this is simulation-only API or this vehicle does not exist.

I'v changed the setting.json

sytelus, 2018-08-23:
You probably don't have correct settings.json that has Car1 as vehicle - just what error says.

EricJin101, 2018-08-24:
After you changed the settings.json, you need to stop the simulation and restart it....

closed 2018-08-24 by EricJin101