microsoft/AirSim #1343: MavLinkMsg/LogViewer direction information

TastiiVegas, 2018-08-22:
Dear AirSim-Team,
I connected an Pixhawk to the simulation via HITL and started the MavLinkTestApp. Then I wanted to analyze the messages in rhe LogViewer.
Now I would like to know which message was sent in which direction?!

Is there a way to find it out?

Or can you help me directly with these messages:
actuatorcontrol_target ID 140
altitude ID 141
attitude_quaternion ID 31
attitudetarget ID 83
batterystatus ID 147
commandack ID 77
extendedsys_state ID 245
globalposition_int ID 33
gpsraw_int ID 24
heartbeat ID 0
highresimu ID 105
hilactuator_controls ID 93
localposition_ned ID 32
manualcontrol ID 69
rcchannels_override ID 70
rcchannels ID 65 
servooutput_raw ID 36
statustext ID 253
systemtime ID 2
timesync ID 111
vfrhud_raw ID 74
vibration ID 241

Best Regards,
TastiiVegas

TastiiVegas, 2018-09-07:
@lovettchris Did you have some ideas, how to see the direction in your tool?

lovettchris, 2018-12-18:
I believe these are all the messages received from the drone.  Monitoring the messages sent the other way (from AirSim to Drone) would require intercepting them by setting up a mavlink proxy (which can be done using MavLinkTest.exe).

MSBGit, 2020-06-09:
@lovettchris do you have an example of using MavLinkTest.exe to monitor messages sent from AirSim to the HITL Pixhawk? I tired telling Log Viewer to connect to port 14389 but it showed me the same set of messages that I saw when connecting to port 14388.

I want to see these messages:

- mavlinkcom::MavLinkHilSensor - IMU (accelerometers, gyros, magnetometers and pressure)
- mavlinkcom::MavLinkHilGps - GPS data
- mavlinkcom::MavLinkDistanceSensor - Distance sensors for AGL, e.g. simulated ultrasonic sensor

Do ports 14388 and 14389 talk standard MAVLINK so that I could point `mavsdk_server` at them? I tried that but it never saw any MAVLINK components.

jonyMarino, 2020-10-19:
Issue solved. doc update pending

closed 2020-10-19 by jonyMarino